cortex-lab/phy #985: Error on linux when installing klustakwik2 for clustering

reinderdorman, 2020-03-26:
note: this might be more suitable for kwikteam/klustakwik2, but since that's a bit old I'm not sure if the issue will be picked up?

On a few machines I noticed that, to use the re-clustering option, you need to install klustakwik2. pip install klustakwik2 worked fine on a few windows machines, but when trying it on a linux (lubuntu 16.04, python 3.7) machine I get a pretty large error, which seems to come down to some problems in compute_cluster_masks_cy.c (attached the error in a doc instead of posting here since it's rather long). 

I  fancy the reclustering option somewhat, so any help is appreciated!

[error pip install klustakwik2.docx](https://github.com/cortex-lab/phy/files/4387083/error.pip.install.klustakwik2.docx)

rossant, 2020-04-03:
The log seems to indicate you're using python 3.8? Try to downgrade your environment to 3.7 or even 3.6? Klustakwik2 is indeed old and was developed on an earlier version of Python.

reinderdorman, 2020-04-03:
Yes, that helped! thanks!

To elaborate for anyone looking at this, I downgraded to python 3.7.6 inside the phy2. Had to reinstall some packages, but just running through the standard installation steps worked perfectly.

closed 2020-04-06 by rossant