cortex-lab/phy #986: Phy2 freezzing

llobetv, 2020-03-28:
Hi,
I try to install new phy and I encounter one big issue..., phy freeze one in ten when I select a new cluster and me my whole computer freeze until the timeout is over:

12:00:29.138 [D] supervisor:812       Clusters selected: [160] (140)
12:00:38.221 [E] qt:178               Timeout in _block().
12:00:59.935 [W] actions:213          Error when executing action Next best.
12:00:59.975 [D] actions:214          Traceback (most recent call last):
  File "/home/llobet/Documents/librairie/phy/phy/gui/actions.py", line 211, in wrapped
    return callback(*args)
  File "/home/llobet/miniconda3/envs/phy2/lib/python3.8/site-packages/phylib/utils/event.py", line 141, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/llobet/Documents/librairie/phy/phy/cluster/supervisor.py", line 847, in _on_action
    _block(lambda: not self._is_busy)
  File "/home/llobet/Documents/librairie/phy/phy/gui/qt.py", line 182, in _block
    raise RuntimeError("Timeout in _block().")
RuntimeError: Timeout in _block().

Have you an idea about the origin of the problem?

Best regards,
Victor Llobet

llobetv, 2020-03-28:
When I increased timeout I got this: 

/home/llobet/Documents/librairie/phy/phy/cluster/views/correlogram.py:108: UserWarning: Persisting input arguments took 23.07s to run.
If this happens often in your code, it can cause performance problems 
(results will be correct in all cases). 
The reason for this is probably some large input arguments for a wrapped
 function (e.g. large strings).
THIS IS A JOBLIB ISSUE. If you can, kindly provide the joblib's team with an
 example so that they can fix the problem.
  ccg = self.correlograms(self.cluster_ids, self.bin_size, self.window_size)

rossant, 2020-03-28:
Thanks for the details. This is really curious. Could you perhaps edit this correlogram.py file line 108 and add:

`print(self.cluster_ids, self.bin_size, self.window_size)`

llobetv, 2020-03-28:
[64, 197] 0.0003 0.05
/home/llobet/Documents/librairie/phy/phy/cluster/views/correlogram.py:109: UserWarning: Persisting input arguments took 6.00s to run.
If this happens often in your code, it can cause performance problems 
(results will be correct in all cases). 
The reason for this is probably some large input arguments for a wrapped
 function (e.g. large strings).
THIS IS A JOBLIB ISSUE. If you can, kindly provide the joblib's team with an
 example so that they can fix the problem.
  ccg = self.correlograms(self.cluster_ids, self.bin_size, self.window_size)

rossant, 2020-03-28:
Indeed... I don't understand why it takes a huge time (several seconds) to persist this input.
Does this persist if you launch the GUI with `--clear-cache --clear-state` at the end of the `phy...` command?

llobetv, 2020-03-28:
It doesn't change anything... ccg = self.correlograms(self.cluster_ids, self.bin_size, self.window_size) still take age to run randomly (sometimes 60s)

rossant, 2020-03-28:
What is your OS, RAM, hard drive characteristics (the one where the data is stored), its total size and free size? What is the size of the .phy subdirectory inside the data directory?

llobetv, 2020-03-28:
Ubuntu 18.04
16 Gb
1To SSD, nearly empty
.phy/ size: 78.4Mo

rossant, 2020-03-28:
Hmm. Go to `phy/apps/base.py` and comment line 822 (`'_get_correlograms'`)
Is it still slow?

llobetv, 2020-03-28:
I found the issue, it was my fault...

I connect my ssd to an usb switch which should create bug when their was disk access... now that i connect it directly to my computer everything work well.

Sorry for that and thank you very much for your help.
Best,
Victor

rossant, 2020-03-28:
good to know!

closed 2020-03-28 by rossant

TomBugnon, 2021-06-30:
@rossant could we please reopen this?

I regularly have the same issue:
[image]
[image]

And sometimes (but not always) get the joblib message
```
The reason for this is probably some large input arguments for a wrapped
function (e.g. large strings).
THIS IS A JOBLIB ISSUE. If you can, kindly provide the joblib's team with an
example so that they can fix the problem.
```

This happens sparsely most of the time but now I'm curating a dataset with a massive number of spikes and I can't change the group of any cluster because of this...

I'm working on a server with huge RAM and CPU specs so I don't think that's the issue