cortex-lab/phy #987: phy GUI has invisible parts

HunsonAbadeer1, 2020-03-30:
I am trying to vie my kilosort data in the phy GUI, but only the ClusterView and the SimilarityView contain anything. The ClusterScatterView seems o contain invisible information, if I click on it randomly it changes the ClusterView and the SimilarityView. Does anyone have an idea how to solve this.
[image]
There are no error messages when I start it with the anaconda prompt. I tried it with the clear state as well, that didn´t change anything. The Graphics Card is an NVIDIA GeForce RTX 2080 Ti, the driver version is 26.21.14.4166.
With the debug command I get a lot of these:
[image]
Any idea how to fix this?

chouran, 2020-03-30:
I got the exact same errors with SpyKing Circus data. Did you manage to fix it ?

HunsonAbadeer1, 2020-04-02:
Nope, so far not. I have also written to other people with the same problem and there is so far no solution.

rossant, 2020-04-03:
Is this Windows? It might be that the Python process is using, for some reason, the native OpenGL 1.x driver (which is extremely old) rather than the NVIDIA driver. I'll try to see if I can reproduce the issue on my Windows computer.

HunsonAbadeer1, 2020-04-03:
Yes, this is Windows.

chouran, 2020-04-03:
If it might help, there is a problem with the windows PyOpenGL package, apparently, some files are missing. I have follow this [tutorial](https://codeyarns.com/2012/04/27/pyopengl-installation-notes-for-windows/ ) and download and install the unofficial package supposed to fix those bug (from [here](https://www.lfd.uci.edu/~gohlke/pythonlibs/#pyopeng). However there is now a problem with the pyrender package which is said not to be compatible with this new version of PyOpenGL. It might provide some insight

rossant, 2020-04-03:
Thanks that's helpful!
Otherwise just "pip install pyopengl" is not working on Windows?

RynzzZ, 2020-04-03:
Hi all, 

I'm currently having the same probelms. But I'm connecting to lab PC thru remote desktop due to current intense COVID-19 situation. For me, this Phy GUI having invisible parts problem only started to show up after connecting remotely and has never showed up while working on-stie. However, other softwares, like MATLAB, seem working properly with remote desktop, including all the plotting functions I have used so far.  

Am I getting this problem because of remote desktop microsoft due to some python packages/drivers used by Phy2 imcompatible with/don't support remote desktop?  Or is this a new problem that came up only recenly due to other reasons irrelevant to working remotely?

Any suggestions/fixes are deeply appreciated! Thanks!

HunsonAbadeer1, 2020-04-06:
I tried both as well, with the unoffical PyOpenGL i have the problem that none of these versions is a supported wheel on the platform, probably since i am using version 2.7. Is there a usable version for me or do I rather upgrade phy?

rossant, 2020-04-07:
Python 2.7 is unsupported, you should use a Python 3.6+ environment

HunsonAbadeer1, 2020-04-07:
I changed to 3.8.2 and know was ablto follow chourans advise, but that did not change anything. The result is still identical.

chouran, 2020-04-09:
I think the issue might come from vispy directly, I also have this [issue](https://github.com/vispy/vispy/issues/1342)

rossant, 2020-04-10:
In case that's useful, phy is working correctly on my Windows 10 machine with the following versions:

```
Python 3.7.3

PyOpenGL                           3.1.0
PyQt5                              5.12.1
PyQt5-sip                          4.19.15
PyQtWebEngine                      5.12.1
```

chouran, 2020-04-10:
Thank you @rossant, i'll try to downgrade my packages. Do you have a dedicated or integrated GPU ? One of the vispy developers pointed that integrated Intel GPU might be the core of the problem

rossant, 2020-04-10:
@chouran absolutely, Intel GPUs cause some problems. But a recent Intel graphics driver seems to have fixed it : https://github.com/cortex-lab/phy/issues/957#issuecomment-609498355
I have an nvidia GPU on my Windows machine, this is the recommended graphics configuration for phy.

HunsonAbadeer1, 2020-04-11:
I tried with the versions that worked for rossant and than with rossants versions and the PyOpenGl Package recommended by chouran, both still had the same result as in the beginning. Did anyone else that had the same problem encounter a solution by now?

chouran, 2020-04-11:
Same for me. I’ve tried every possible combinations of package versions, same results, whatever the installation chosen. The Vispy library also fails to work. The intel drivers for some types of old integrated graphic card are most likely the core of the problem. The one I have is an Intel Graphics 3000 updated and supporting OpenGL 3.1, however it is no longer patched since 2015.

I’ve tried with a more recent computer, with an Intel graphics 4400, and everything works fine, both phy and vispy ..

HunsonAbadeer1, 2020-04-15:
I was able to use the Phy GUI on the same Dataset with just the very basic manual from GitHub and no fancy packages on another computer. I think the problem was definetly the  NVIDIA GeForce RTX 2080 Ti, I used an Intel(R) UHD Graphics 630 and it worked on the first try.

closed 2020-06-08 by rossant