cortex-lab/phy #989: Use Phy to compute template similarity in SpikeInterface

alejoe91, 2020-03-31:
Hi @rossant 

In SpikeInterface we have our own version of the `compute_similarity` [function](https://github.com/SpikeInterface/spiketoolkit/blob/master/spiketoolkit/postprocessing/postprocessing_tools.py#L905-L928). @samuelgarcia suggested that it's it's probably better if we have a soft dependency on phy/phylib.

Could you point me to the correct function that I should use?

Cheers,
Alessio

rossant, 2020-04-03:
Hi @alejoe91 , there is no good such function in phy at the moment, phy uses the similar_templates.npy matrix computed by kilosort2

samuelgarcia, 2020-04-03:
Hi Cyril.
How do you refresh this value when you process a merge or a split  inside phy ?

rossant, 2020-04-03:
It considers the similarity between the templates from which the clusters originate from (the ones with the most spikes from each cluster). Certainly not a good idea solution and this should be improved eventually. [Note that there's a plugin example showing how to implement another similarity function](https://phy.readthedocs.io/en/latest/plugins/#writing-a-custom-cluster-similarity-metrics).

alejoe91, 2020-04-03:
Hi @rossant 

Thanks for the clarification. Do you know how KS implements such similarity? At the moment we use the correlation coefficient, but I think that's not really good as it doesn' consider differences in amplitude, for example. 

As the similarity between templates is cheap to compute, maybe it would be a better option to have it centralized in phy? So you could recompute it after merges/splits... What do you think?

rossant, 2020-04-03:
Yes that would make sense. I think the KS implementation of the similarity matrix is currently done on the GPU using a custom CUDA kernel. For the integration in phy I would prefer if we could have a pure NumPy version, don't know if that's feasible. Looping in @marius10p

alejoe91, 2020-04-03:
For reference, here is the [SI implementation](https://github.com/SpikeInterface/spiketoolkit/blob/master/spiketoolkit/postprocessing/postprocessing_tools.py#L1025-L1048) (line numbers have changed from the link shared before)