cortex-lab/phy #990: WaveformView displays noise

MarinaSlashcheva, 2020-04-01:
Hi,

Phy WaveformView displays weird (line) noise in each cluster. I first thought that my data is screwed up or it is the problem of referencing (reference to the tip of Neuropixels probe), but TraceView with raw data looks fine (and has spikes), raw data in Kilosort GUI also looks fine. 

In the table with all the clusters they also appear normal, as well as FeatureView, amplitude, correlogram, etc (screenshots attached). I have an impression it is a problem of displaying cluster Waveforms (and the problem persists when all traces are shown or if they are averaged).

I tried to delete state.json file both locally (in data folder) and in home folder, it did not help.

Thanks for help!
[image]
[image]
[image]

rossant, 2020-04-03:
Hi, would you have a way to send me your data, via dropbox, google drive, or similar? It would be very helpful for debugging. Thanks!

MarinaSlashcheva, 2020-04-07:
Hi Cyrille, I uploaded the data to the google drive, it is already spike sorted but has a raw binary file as well. I hope I uploaded all necessary files, let me know if I forgot something https://drive.google.com/file/d/1igs5rPPQDXG2_CMULfUpy4o3ayhxFyZi/view?usp=sharing Best regards,Marina Slashcheva From: Cyrille RossantSent: Friday, April 3, 2020 11:05 AMTo: cortex-lab/phyCc: Marina Slashcheva; AuthorSubject: Re: [cortex-lab/phy] WaveformView displays noise (#990) Hi, would you have a way to send me your data, via dropbox, google drive, or similar? It would be very helpful for debugging. Thanks!

MarinaSlashcheva, 2020-04-07:
Hi Cyrille, 

I sent you an email with the link

EmmettJT, 2021-01-14:
Hi Cyrille,
 
I also have this same issue: 
[image]
Any update on what is causing it and whether there is a way to fix it? 

Thanks!

EmmettJT, 2021-01-14:
If I don't open Phy with my raw data (I changed my raw data path in params.py to point at an empty dir) I can see nice waveforms. 
[image]

anoushkajain, 2021-08-05:
Hi Cyrille,

I am also facing have this same issue:

[image]

Are there any updates to resolve? 

Best,
Anoushka

Zhentaozha, 2024-03-09:
Hi 
I am also facing have this same issue in 2024
[image]