cortex-lab/phy #991: AssertionError:   assert x.shape[1] == mat.shape[0] AssertionError

catubc, 2020-04-02:
Hi. The YASS team is trying generate data for phy2, but seems there's an additional check being done that is now failing.  Can you please clarify what this error is?
```

(phy2) cat@cat-System-Product-Name:~/Downloads/yass/samples/10chan/phy$ phy template-gui params.py
09:59:37.961 [W] model:1153           File data.bin does not exist.
09:59:37.997 [W] model:158            Path /home/cat/Downloads/yass/samples/10chan/phy/data.bin does not exist, trying ephys.raw filename.
09:59:37.997 [W] model:161            Error while loading data: File `/home/cat/Downloads/yass/samples/10chan/phy/ephys.raw.bin` not found.
09:59:38.071 [E] __init__:62          An error has occurred (AssertionError): 
Traceback (most recent call last):
  File "/home/cat/anaconda3/envs/phy2/bin/phy", line 8, in <module>
    sys.exit(phycli())
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/template/gui.py", line 199, in template_gui
    gui = controller.create_gui()
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/base.py", line 1638, in create_gui
    self.supervisor.attach(gui)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 941, in attach
    self._create_views(
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 760, in _create_views
    gui, data=self.cluster_info, columns=self.columns, sort=sort)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 916, in cluster_info
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 916, in <listcomp>
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 745, in get_cluster_info
    out[key] = func(cluster_id)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/base.py", line 1150, in get_best_channel_label
    return self._get_channel_labels([self.get_best_channel(cluster_id)])[0]
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/utils/context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/base.py", line 1144, in get_best_channel
    channel_ids = self.get_best_channels(cluster_id)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/utils/context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/template/gui.py", line 149, in get_best_channels
    template = self.model.get_template(template_id)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phylib/io/model.py", line 855, in get_template
    return self._get_template_dense(template_id, channel_ids=channel_ids)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phylib/io/model.py", line 796, in _get_template_dense
    template = self._unwhiten(template_w).astype(np.float32)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phylib/io/model.py", line 688, in _unwhiten
    assert x.shape[1] == mat.shape[0]
AssertionError

```

rossant, 2020-04-03:
Hi. Do you confirm that you're saving the templates in dense and not sparse format? templates.npy should have the shape (n_templates, n_samples, n_channels). The inverse whitening matrix should have the shape (n_channels, n_channels), otherwise this assertion fails.

catubc, 2020-04-03:
The templates are full matrices, but we don't use a whitening matrix for anything (or save one).  Is there a way to bypass that matrix?  Otherwise, what is the filename of the matrix? We can just make a dummy one.
Thanks!

rossant, 2020-04-03:
It should not fail like this in the absence of a whitening matrix, so that's definitely a bug. In the meantime creating a dummy whitening_mat.npy should work.

catubc, 2020-04-06:
Hi. I created a whitening_mat.npy file with size [n_channels, n_channels].  I am still getting the same error and it's not clear what the assertion error refers to.

```
(phy2) cat@cat-System-Product-Name:~/Downloads/yass/samples/10chan/phy$ phy template-gui params.py
09:55:14.695 [E] __init__:62          An error has occurred (AssertionError): 
Traceback (most recent call last):
  File "/home/cat/anaconda3/envs/phy2/bin/phy", line 8, in <module>
    sys.exit(phycli())
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/template/gui.py", line 199, in template_gui
    gui = controller.create_gui()
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/base.py", line 1638, in create_gui
    self.supervisor.attach(gui)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 941, in attach
    self._create_views(
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 760, in _create_views
    gui, data=self.cluster_info, columns=self.columns, sort=sort)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 916, in cluster_info
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 916, in <listcomp>
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/cluster/supervisor.py", line 745, in get_cluster_info
    out[key] = func(cluster_id)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/base.py", line 1150, in get_best_channel_label
    return self._get_channel_labels([self.get_best_channel(cluster_id)])[0]
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/utils/context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/base.py", line 1144, in get_best_channel
    channel_ids = self.get_best_channels(cluster_id)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/utils/context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phy/apps/template/gui.py", line 149, in get_best_channels
    template = self.model.get_template(template_id)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phylib/io/model.py", line 855, in get_template
    return self._get_template_dense(template_id, channel_ids=channel_ids)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phylib/io/model.py", line 796, in _get_template_dense
    template = self._unwhiten(template_w).astype(np.float32)
  File "/home/cat/anaconda3/envs/phy2/lib/python3.8/site-packages/phylib/io/model.py", line 688, in _unwhiten
    assert x.shape[1] == mat.shape[0]
AssertionError

```

rossant, 2020-04-06:
Any way you could send me a failing example dataset (only the npy files should be enough)?

catubc, 2020-04-06:
Yes, here's the link to the phy directory including the (very small) data.bin file:

https://www.dropbox.com/s/t7q0q244pnudg5g/phy.zip?dl=0

rossant, 2020-04-06:
Thanks. It looks like templates.npy has shape (10, 31, 101) but it should be (n_templates, n_samples, n_channels) whereas you seem to have 10 channels. Some of your arrays should be transposed perhaps?

catubc, 2020-04-06:
Thanks, you were correct (someone changed our template shape and I didn't notice it).

Also to confirm (once I fixed the above issue) I did NOT require making a whitening_mat.npy file.

closed 2020-04-06 by catubc