cortex-lab/phy #992: No module named 'phy.utils.tempdir'

Pascal-IINS, 2020-04-08:
Hello,
I have tried to install re-install several times Phy2 on a clean anaconda on an admin session. When I open the Phy2 template-gui, the prompt tells:
No module named 'phy.utils.tempdir'
Below is the Traceback...

15:49:46.981 [W] model:1192           File ..\3C030-S18_191208.dat does not exist.
15:49:47.206 [W] model:513            Unreferenced clusters found in templates (generally not a problem)
15:49:47.499 [W] model:535            Unreferenced clusters found in spike_clusters (generally not a problem)
15:49:47.629 [W] model:577            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
15:49:47.641 [W] traces:454           File F:\Kilosort\Cheeseboard\3C030\3C030-S18_191208\3C030-S18_191208.dat does not exist.
15:49:47.645 [W] traces:454           File F:\Kilosort\Cheeseboard\3C030\3C030-S18_191208\3C030-S18_191208.dat does not exist.
15:49:49.807 [E] plugin:111           No module named 'phy.utils.tempdir'
Traceback (most recent call last):
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\utils\plugin.py", line 109, in discover_plugins
    mod = imp.load_module(modname, file, path, descr)  # noqa
  File "c:\programdata\anaconda3\envs\phy2\lib\imp.py", line 234, in load_module
    return load_source(name, filename, file)
  File "c:\programdata\anaconda3\envs\phy2\lib\imp.py", line 171, in load_source
    module = _load(spec)
  File "<frozen importlib._bootstrap>", line 702, in _load
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\teamROUXadmin\.phy\plugins\recluster.py", line 12, in <module>
    from phy.utils.tempdir import TemporaryDirectory
ModuleNotFoundError: No module named 'phy.utils.tempdir'
15:49:59.810 [W] plugin:151           The plugin Recluster couldn't be found.
15:50:01.752 [E] base:1739            'GUI' object has no attribute 'edit_actions'
Traceback (most recent call last):
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\gui\qt.py", line 532, in run
    result = self.fn(*self.args, **self.kwargs)
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 175, in _worker
    self.on_select(cluster_ids=cluster_ids, **kwargs)
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 143, in on_select
    self.plot(**kwargs)
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\feature.py", line 369, in plot
    self._plot_points(bunch, clu_idx=clu_idx)
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\feature.py", line 229, in _plot_points
    assert px.data.shape == py.data.shape
AssertionError
Traceback (most recent call last):
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\gui\qt.py", line 532, in run
    result = self.fn(*self.args, **self.kwargs)
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 175, in _worker
    self.on_select(cluster_ids=cluster_ids, **kwargs)
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 143, in on_select
    self.plot(**kwargs)
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\feature.py", line 369, in plot
    self._plot_points(bunch, clu_idx=clu_idx)
  File "c:\programdata\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\feature.py", line 229, in _plot_points
    assert px.data.shape == py.data.shape
AssertionError

rossant, 2020-04-08:
There seems to be incompatible plugins at C:\Users\teamROUXadmin.phy\plugins\, could you rename this directory to `plugins~` and try again? Let me know if you need help to upgrade the plugins.
You could also try to run phy with the `--clear-cache` and `--clear-state` options on the command line

Pascal-IINS, 2020-04-09:
Thanks ! The error pop-up message disappeared, although warnings persist in the anaconda prompt:

13:11:32.160 [W] plugin:151           The plugin Recluster couldn't be found.
13:11:32.161 [W] plugin:151           The plugin SplitShortISI couldn't be found.

I also needed to trash the .phy folder from the data directory for the template-gui to work.
Do you advice to reinstall the plugins via pip ?

rossant, 2020-04-09:
Where do the plugins come from? Are they on GitHub? If so I could provide advice to upgrade them.

Pascal-IINS, 2020-04-09:
For Phy, I used buzsaki lab plugins. But since i upgraded to Phy2 I installed all using:
pip install colorcet pyopengl qtconsole requests traitlets tqdm joblib click mkdocs
pip install git+https://github.com/cortex-lab/phy.git@dev
pip install git+https://github.com/cortex-lab/phylib.git

Pascal-IINS, 2020-04-09:
I am really sorry, there is still a glitch when I cluster cut (k), the waveformView does not update correctly, showing the newly created cluster's average waveform in red exactly overlapping the former one. Would you have any suggestion on how to fix that ?

rossant, 2020-04-09:
does this problem persist if you save, exit, and reopen phy?

Pascal-IINS, 2020-04-09:
Indeed. It looks like the average waveforms are not updated, neither after cutting a cluster nor vice-versa after merging 2 clusters (it shows both average waveforms in blue and grey).

rossant, 2020-04-09:
Also if you delete the .phy subdirectory inside your data directory?
Could you post a screenshot before/after the split?

rossant, 2020-04-09:
BTW do you have the raw data? Do you see the raw spike waveforms, or the template waveforms? If the latter, I think it's normal that the template waveforms are not updated when splitting.

Pascal-IINS, 2020-04-09:
Here it is before:
[image]

And after pressing 'k':
[image]

Do you mean the raw.dat file ? I had removed it at some point to gain space on my disk, and put it back, realizing it could be a problem while clustering. It did not change the issue. The strange thing is that last week I didn't experience the issue with the waveforms.

rossant, 2020-04-09:
ok thanks, this is what I thought. The waveform view shows the templates (see at the bottom of that view). Press W to switch to the raw waveforms or mean waveforms, but if the raw data is unavailable, that won't work. Templates do not change when splitting.

Pascal-IINS, 2020-04-10:
It was so simple that it is quite embarrassing. Thank you for your feedback !

closed 2020-06-08 by rossant

dervinism, 2025-05-15:
Not sure if anyone is following this
@rossant @Pascal-IINS

dervinism, 2025-05-19:
The line could be commented out, and then the plugin would load. But there are other lines in the plugin code that may potentially cause error. There was clearly a change in the Phy API.