cortex-lab/phy #993: Problem with CCGs vizualization

LisaRoux5297, 2020-04-09:
Hi,

We are facing an issue with Phy2 when vizualizing CCGs (which we use extensively to decide whether or not we should merge clusters, in addition to the similarity score): for some pairs of neurons, the CCG is extremely small and it makes it impossible to evaluate the refractory period. The 'normalization' function doesn't seem to solve the problem in some cases). 

[Exemple normalisation Phy.pdf](https://github.com/cortex-lab/phy/files/4458682/Exemple.normalisation.Phy.pdf)
I am attaching exemples of 2 pairs of units with their CCGs, before and after pressing the N key for normalization. All units in these exemples fire throughout the entire recording session. 

Is there any way this problem can be solved?

Thank you very much for your help,
Lisa

rossant, 2020-04-10:
Are you saying that it would make more sense for the normalization option to use an independent scaling for each subplot, rather than for each row?

LisaRoux5297, 2020-04-10:
It would indeed probably help in the types of cases we are encountering. 

When we use CCGs and ACGs for spike sorting, we never really pay attention to the "heights" of the histograms and do not compare these heights between ACG and CCGs. Being able to vizualize the refractory periods is however absolutly key. It would really help if the CCG scale could be adjusted in order to make this inspection possible.

LisaRoux5297, 2020-05-01:
Sorry for bothering you with this again but do you think this feature (independent scaling of each subplot) could be implemented?

rossant, 2020-05-04:
Yes I'll do it as soon as possible

rossant, 2020-06-08:
this should be fixed on the master branch, please reopen if not

closed 2020-06-08 by rossant