cortex-lab/phy #996: dore dumped after QGLXContext: Failed to create dummy context

JoseGuzman, 2020-04-16:
I'm using Ubuntu 18.04.4 and found the error indicated below. I installed the **PyQtWebEngine**  with pip as suggested. I can reproduce the error in a freshly created conda environment.

I'm running Phy via `ssh -X`, but it was not a problem until now.

```bash
QGLXContext: Failed to create dummy context
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
Aborted (core dumped)```

rossant, 2020-04-20:
Running phy via ssh is not currently supported at the moment, unfortunately. However I'm considering ways to make it work in the medium term.

JoseGuzman, 2020-04-20:
good to know, I also think using Phy via ssh has not the highest priority!

claireward, 2021-07-26:
Hi @JoseGuzman and @rossant,
I am getting this same error message on my linux machine. I'm not certain on how to get around it. Do you have any updates or more details on how to get Phy2 up and running?

claireward, 2021-07-26:
> Hi @JoseGuzman and @rossant,
> I am getting this same error message on my linux machine. I'm not certain on how to get around it. Do you have any updates or more details on how to get Phy2 up and running?

I was able to resolve this issue, however, I 'm not exactly certain what did the trick. I uninstalled, rebooted my machine, and then installed again and everything was fine.