cortex-lab/phy #997: Unable to add IPythonView to GUI

mswallac, 2020-04-19:
**The issue:** 

Running: ``` phy template-gui params.py``` works just fine. This error occurs when attempting to add IPythonView to the GUI through View > Add IPythonView.

I read in a possibly related issue that this may be related to python 3.7 vs. 3.8? (Source: [https://github.com/jupyter/qtconsole/issues/400](https://github.com/jupyter/qtconsole/issues/400)). Let me know what you think!

I am on Windows 10.

I'd like to use the IPythonView to run analysis scripts, so if you know of any other way to run .py files from within phy, that would also help.

**I have tried:**
- Switching to python 3.7.7 (results in another error -- no surprise since it's not the version currently used with phy)
- Recreating anaconda environment several times according to the installation instructions

Here is the excerpt of the relevant error:
```bash
[90m16:38:50.647 [D] actions:214          Traceback (most recent call last):
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 528, in get
    value = obj._trait_values[self.name]
KeyError: 'iopub_socket'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 528, in get
    value = obj._trait_values[self.name]
KeyError: 'iopub_thread'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\black\desktop\phytools\phy\phy\gui\actions.py", line 211, in wrapped
    return callback(*args)
  File "c:\users\black\desktop\phytools\phy\phy\gui\gui.py", line 669, in create_and_add_view
    view = fn()
  File "c:\users\black\desktop\phytools\phy\phy\apps\base.py", line 1542, in create_ipython_view
    view.start_kernel()
  File "c:\users\black\desktop\phytools\phy\phy\gui\widgets.py", line 45, in start_kernel
    self.kernel_manager.start_kernel(show_banner=False)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\inprocess\manager.py", line 46, in start_kernel
    self.kernel = InProcessKernel(parent=self, session=self.session)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 72, in __init__
    super(InProcessKernel, self).__init__(**traits)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\ipkernel.py", line 71, in __init__
    self.shell.displayhook.pub_socket = self.iopub_socket
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 556, in __get__
    return self.get(obj, cls)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 535, in get
    value = self._validate(obj, dynamic_default())
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 67, in _default_iopub_socket
    return self.iopub_thread.background_socket
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 556, in __get__
    return self.get(obj, cls)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 535, in get
    value = self._validate(obj, dynamic_default())
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 59, in _default_iopub_thread
    thread = IOPubThread(self._underlying_iopub_socket)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\iostream.py", line 71, in __init__
    self._setup_event_pipe()
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\iostream.py", line 90, in _setup_event_pipe
    self._event_puller = ZMQStream(pipe_in, self.io_loop)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\zmq\eventloop\zmqstream.py", line 127, in __init__
    self._init_io_state()
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\zmq\eventloop\zmqstream.py", line 546, in _init_io_state
    self.io_loop.add_handler(self.socket, self._handle_events, self.io_loop.READ)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\tornado\platform\asyncio.py", line 100, in add_handler
    self.asyncio_loop.add_reader(fd, self._handle_events, fd, IOLoop.READ)
  File "C:\Users\black\anaconda3\envs\phy2\lib\asyncio\events.py", line 501, in add_reader
    raise NotImplementedError
NotImplementedError

[90m16:38:50.647 [D] actions:214          Traceback (most recent call last):
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 528, in get
    value = obj._trait_values[self.name]
KeyError: 'iopub_socket'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 528, in get
    value = obj._trait_values[self.name]
KeyError: 'iopub_thread'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\users\black\desktop\phytools\phy\phy\gui\actions.py", line 211, in wrapped
    return callback(*args)
  File "c:\users\black\desktop\phytools\phy\phy\gui\gui.py", line 669, in create_and_add_view
    view = fn()
  File "c:\users\black\desktop\phytools\phy\phy\apps\base.py", line 1542, in create_ipython_view
    view.start_kernel()
  File "c:\users\black\desktop\phytools\phy\phy\gui\widgets.py", line 45, in start_kernel
    self.kernel_manager.start_kernel(show_banner=False)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\inprocess\manager.py", line 46, in start_kernel
    self.kernel = InProcessKernel(parent=self, session=self.session)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 72, in __init__
    super(InProcessKernel, self).__init__(**traits)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\ipkernel.py", line 71, in __init__
    self.shell.displayhook.pub_socket = self.iopub_socket
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 556, in __get__
    return self.get(obj, cls)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 535, in get
    value = self._validate(obj, dynamic_default())
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 67, in _default_iopub_socket
    return self.iopub_thread.background_socket
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 556, in __get__
    return self.get(obj, cls)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\traitlets\traitlets.py", line 535, in get
    value = self._validate(obj, dynamic_default())
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 59, in _default_iopub_thread
    thread = IOPubThread(self._underlying_iopub_socket)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\iostream.py", line 71, in __init__
    self._setup_event_pipe()
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\ipykernel\iostream.py", line 90, in _setup_event_pipe
    self._event_puller = ZMQStream(pipe_in, self.io_loop)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\zmq\eventloop\zmqstream.py", line 127, in __init__
    self._init_io_state()
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\zmq\eventloop\zmqstream.py", line 546, in _init_io_state
    self.io_loop.add_handler(self.socket, self._handle_events, self.io_loop.READ)
  File "C:\Users\black\anaconda3\envs\phy2\lib\site-packages\tornado\platform\asyncio.py", line 100, in add_handler
    self.asyncio_loop.add_reader(fd, self._handle_events, fd, IOLoop.READ)
  File "C:\Users\black\anaconda3\envs\phy2\lib\asyncio\events.py", line 501, in add_reader
    raise NotImplementedError
NotImplementedError

```

And the log file just in case:
[phy.log](https://github.com/cortex-lab/phy/files/4500178/phy.log)

rossant, 2020-04-20:
I would have thought that Python 3.7 would work. What issues do you have with it exactly? Are you able to set up a fresh conda env with Python 3.7 and all the dependencies? (perhaps you have to edit the environment.yml file and write "python=3.7" or something)

mswallac, 2020-04-20:
Good call! What fixed it was adding =3.7 to the command which initially creates the environment (the environment.yml file seems to already contain the python=3.7 dependency).
I think what was happening was that Anaconda installed Python 3.8 by default when given the original command, which resulted in some incompatible packages being used down the line.

Changed this :
```conda create -n phy2 python pip numpy matplotlib scipy scikit-learn h5py pyqt cython pillow -y```

To this :
```conda create -n phy2 python=3.7 pip numpy matplotlib scipy scikit-learn h5py pyqt cython pillow -y```

closed 2020-04-20 by mswallac