cortex-lab/phy #998: DOCs or --help should be improved to get a dumb user going or pypi upload fixed up?

yarikoptic, 2020-04-21:
In a sense it is a duplicate of elderly #775.  IMHO there is still a usability issue which ideally should be addressed.

I am not new to Python but I am brand new to phy.  

I loosely followed README.md and did 
```shell
$ conda install pip numpy matplotlib scipy scikit-learn h5py pyqt cython pillow
$ pip install phy  # since hey -- I know Python and I know that I could try to get it from pypi
#  would be nice to have it in conda too
```
to install it.  Then progressed to the next stage in README.md:
```
$> cd yoh-klusta/phy/         
params.py  recording.dat

$> cat params.py 
dat_path =r'recording.dat'
n_channels_dat = 19
dtype = 'float64'
offset = 0
sample_rate = 2000.0
hp_filtered = False%  

$> phy template-gui params.py           
Usage: phy [OPTIONS] COMMAND [ARGS]...

Error: No such command 'template-gui'.
```
so as you can see -- there is params.py, ...

`--help` was not really helpful to get me unstuck (I expected it to list the COMMANDs it has available, like e.g. `git --help` does):
```shell
$> phy --help
Usage: phy [OPTIONS] COMMAND [ARGS]...

  By default, the `phy` command does nothing. Add subcommands with plugins
  using `attach_to_cli()` and the `click` library.

Options:
  --version   Show the version and exit.
  -h, --help  Show this message and exit.

$> phy --version
phy, version 1.0.9
```
so I was left wondering either it is an installation issue, or documentation, or me or all of the above.

here it seems to be some installation (or upload to pypi) issue since
```
$> python -c 'import phy; print(phy.__version__)'
1.0.9

$> python -c 'import phy.apps'                   
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'phy.apps'
```
so there is `phy` but no `phy.apps`.

FWIW, if it is an installation issue (needed to install some additional components), in DataLad for commands we are aware about but "extension" for which is not installed, we provide a hint, e.g.:
```shell
$> datalad containers-run                                          
datalad: Unknown command 'containers-run'.  See 'datalad --help'.

Hint: Command containers-run is provided by (not installed) extension datalad-container.
```
which you might like to adopt in case it is some "optional installation" issue.

rossant, 2020-04-27:
Sorry about that, the current version on pypi is outdated, I was going to release the 2.0 version before the outbreak and didn't get a chance to finalize it yet. Doing `pip install phy --pre` should work though

yarikoptic, 2020-04-27:
Thank you @rossant - I will try the prerelease.