cortex-lab/phy #1000: ImportError: cannot import name 'SpikeSelector' from 'phylib.io.array'

zuhayousuf, 2020-04-29:
I installed phy, and when I ran 'phy template-gui params.py' this is the error I got. Any advice on how to resolve this?? Any help appreciated!!!

rossant, 2020-04-29:
might be a version conflict, uninstall phy completely and reinstall the latest version following the [instructions](https://github.com/cortex-lab/phy#installation-instructions)

zuhayousuf, 2020-04-30:
How do we completely uninstall phy from the terminal? Is there a specific command we can use?

zuhayousuf, 2020-04-30:
When I tried to use 'conda uninstall phy2', this is what I got (Image attached). Does this seem like phy has been uninstalled successfully? And now I can reinstall phy2?
[image]

zuhayousuf, 2020-04-30:
I completely uninstalled phy and reinstalled using the instructions provided, but it's giving me an OSError again

zuhayousuf, 2020-04-30:
Okay so after I reinstalled Anaconda, it seems to work. Thanks!

closed 2020-04-30 by zuhayousuf

TRuikes, 2020-04-30:
Had the same problem; even after a clean install according to instructions. It might be because phylib is installed through [pipy](https://pypi.org/project/phylib/) and that one is outdated?

afonanar, 2021-09-03:
I am having a similar issue but with ´selector´ and not ´spikeselector´ and trying to follow the posted suggestions did not solved the problem. 
Also everything was working fine before, but some how it is not working anymore. 
If anyone has a suggestion I would appreciate a lot...

afonanar, 2021-09-08:
> I had a similar issue after updating to the latest beta version of phy2. It seems that phylib was not updated in the process. I was abple to fix it by updating phylib through PyPI. On the phy2 environment I ran:
> `python3 -m pip install --upgrade phylib`

Thank you. It is working now