cortex-lab/phy #999: Unable to open spike_clusters.npy or params.phy after installing Phy

zuhayousuf, 2020-04-29:
So I followed the instructions on the Github repo for installing Phy up till here:
phy template-gui params.py

However, when I enter the line above, I get the following error in the picture attached. It is an OSError. Is there any way I can fix this? Any advice appreciated!

[image]

zuhayousuf, 2020-04-29:
Some more comments: my raw flat binary file along with spike_clusters.npy and params.py are in the same folder as they're supposed to be. I'm still confused why it isn't working

rossant, 2020-04-29:
spike_samples.npy is there too?

zuhayousuf, 2020-04-29:
Not spike_samples.npy, but it has spike_times.npy

zuhayousuf, 2020-04-30:
Okay, so the error on my end was that spike_times was not saved. Once I did that, it works. Thanks!

closed 2020-04-30 by zuhayousuf