cortex-lab/phy #1001: Computer freezes on exit

yagui, 2020-04-30:
Hi

This is probably a problem in my computer but I'm not really sure how to debug it.
Sometimes, when I exit phy, the computer freezes completely. I'm using Linux Mint with an Nvidia GeForce GTX 1060, 16 GB RAM and dataset on a SSD.

If I open phy and close it in 5-10 minutes, working in one or two units it's less likely for the computer to freeze, almost never.
If I have been working for half an hour, phy uses almost all memory and it's more likely to freeze on exit, more than 50% of the times.
What I can see is that while using phy, memory usage increases slowly, so I though that it was related to that but even if I try to close phy before it gets to full, and start over, its ~50% chances that it will freeze. If I forget to do it for a while, its very likely to happen.

I ran `phy template-gui params.py --debug 2>>debug.txt`
I attach this file [debug.txt](https://github.com/cortex-lab/phy/files/4561111/debug.txt). The first time it did not freeze and I was working for 20 minutes.
The second time (starting at line 472) it did freeze and I was working for 11 minutes.
From the debug file, what I can see is that the exit is completed in both cases, at least for any debug messages.
When freezes, nothing works (not even Ctrl+Alt+F1, which almost never fails). What it does work beside the power button, is ALT+SYSRQ+ R E I S U B, that is a gently why to reset the computer, so it is not completely dead.
Again, I'm not sure if anyone else is having a similar problem, I couldn't find any reference but the computer only freezes when exiting phy. It is really strange.

Any idea will be appreciated. The good thing is that always everything is correctly saved, but is a bit annoying having to reset every 10-15 minutes.

yagui, 2020-04-30:
Also, this started to happen to me a few months ago, after an update, but I did not tracked well which commit could have done this. Maybe I can try to add more debug message to see where it crashes, but I'm not where to start

nippoo, 2020-04-30:
Can you check if there are any relevant logs in your dmesg / system log? You may want to grep for OOM to see if the out-of-memory killer is killing some bit of the X subsystem for example.

Also, if you know the oldest date where this occurred, you could try using git-bisect to try and chase down the relevant commit...

yagui, 2020-05-01:
Looking the system log as suggested I noticed that every time there was a message from the video card
`asynchronous wait on fence NVIDIA:nvidia.prime:95f9 timed out`
Searching a bit in the web I found this related issue
https://github.com/Tk-Glitch/PKGBUILDS/issues/112
and after updating the nvidia driver now it does not freeze anymore.
Thank you so much @nippoo

closed 2020-05-01 by yagui