cortex-lab/phy #1002: Phy TemplateGUI doesn't launch properly (blank window)

RenskeW, 2020-05-01:
Hi everyone,

A few days ago Phy2 stopped functioning properly on my computer. Whenever I try to launch it in Anaconda terminal the TemplateGUI window does appear but it stays completely blank, although I still can select clusters (they are just not visible). Using the --debug option didn't show any errors. Does anyone know what could be causing this problem? I am using Windows 10 and accessing my computer via Teamviewer (which didn't give problems in the past). 

Thanks! 

[image]
[image]
[image]

RenskeW, 2020-05-12:
Could anybody please answer my question?

closed 2020-05-28 by RenskeW

JuliaT311, 2022-05-17:
Did you find a solution because I have the same problem

RenskeW, 2022-05-19:
Sorry @JuliaT311, I forgot what happened in the end...

malinroe, 2022-05-20:
@JuliaT311 I had the same problem when I worked remotely, and it was fixed after we updated some drivers (I can't remember which ones though, but maybe that can help you).