cortex-lab/phy #1009: font size issue

ElKatz, 2020-05-21:
Hi guys, 

Finally tool the leap from Phy1 to 2 and I'm having font size errors. See image below, where labels are very large. This is especially troubling in the cluster & similarity view. 
My display and GPU settings seem fine. Any way I could change this?

[image]

rossant, 2020-06-08:
[does this help?](https://phy.readthedocs.io/en/latest/plugins/#customizing-the-default-font-size)

ElKatz, 2020-06-13:
Hi Cyrille thanks for the response!
I tried it out and despite verifying the the font_size.py is called, the font size does not change on the GUI. Any other ideas?

Thanks, Leor

rossant, 2020-06-14:
The plugin should change the font size in the graphical views, but not in the cluster view. There is no proper way to change the font size in the cluster view at the moment, but something you could try is locate the `phy/gui/widgets.py` on your hard drive and modify [line 112](https://github.com/cortex-lab/phy/blob/master/phy/gui/widgets.py#L112).

Alternatively you could install the development version of phy by cloning the repository and modify the widgets.py file there.

rossant, 2020-06-14:
Actually you could also [try this plugin example](https://phy.readthedocs.io/en/latest/plugins/#customizing-the-styling-of-the-cluster-view) with something like (not tested):

```
    html, body, table {
        font-size: 8pt;
    }
```

ElKatz, 2020-06-18:
Hey thanks, though no go with the widgets. I added print command in there and nothing appeared so I doubt the DEFAULT is getting called.

I couldn't quite figure out what to do with that last plugin example (mt developer fu is not strong)..

rossant, 2020-06-24:
Here are the instructions:

[image]

and use the code below in `~/.phy/plugins/cluster_view_style.py`:

```python
# import from plugins/cluster_view_styling.py
"""Show how to customize the styling of the cluster view with CSS."""

from phy import IPlugin
from phy.cluster.supervisor import ClusterView

class ExampleClusterViewStylingPlugin(IPlugin):
    def attach_to_controller(self, controller):
        # We add a custom CSS style to the ClusterView.
        ClusterView._styles += """
    html, body, table {
        font-size: 8pt;
    }
        """
```

ElKatz, 2020-06-24:
Hi Cyrill, thanks for hanging in there with me!
Still can't get it to work.
I have inserted a 'print' command in each plugin (font_size & cluster_view_styling) and these do work: the command in each plugin results in something printed on screen, when I use c.Plugins.dirs = [r'C:\Users\Admin\.phy\plugins/']" in my phy_config.py.
However, when I use c.TemplateGUI.plugins = ['font_size'] (or [cluster_view_style]), I do not see anything printed on screen and I get the message 
"  [W] plugin:151           The plugin font_size couldn't be found.[0m" 
(or  [W] plugin:151           The plugin cluster_view_styling couldn't be found.[0m)

And of course-- with all these variations, I still can't get the damn text size to reduce in size :)

rossant, 2020-06-25:
what if you add `c.TemplateGUI.plugins = ['ExampleClusterViewStylingPlugin']` in your config file?

ElKatz, 2020-06-29:
still no go. (btw, sorry for my often late responses-- I only get to lab once/twice a week these days..).
I've attached two files for you to take a look at, just to make sure we're on the same page:
- .phy\phy_config
- .phy\plugins\cluster_view_styling

[files.zip](https://github.com/cortex-lab/phy/files/4847571/files.zip)

rossant, 2020-07-01:
My bad, the files are correct but the plugin code itself should rather be (just tested it):

```python

class ExampleClusterViewStylingPlugin(IPlugin):
    def attach_to_controller(self, controller):
        # We add a custom CSS style to the ClusterView.
        ClusterView._styles += """
        td {
            font-size: 6pt !important;
        }
        """
```

czuba, 2020-07-02:
Ok ok...I think I just relived Leor's last few rounds of debugging, and managed to get this working:

## The plugin
Copy & past the following text into a .py file in your main Phy preferences directory [ `~/.phy/plugins` ]
This one plugin will adjust the font size of plot text (axes, labels, etc) and font size in cluster/similarity list(s)
```
# import from ~/.phy/plugins/resizeGuiFonts.py
""" Tune GUI font sizes to your display/preference."""

from phy import IPlugin
from phy.plot.visuals import TextVisual
from phy.cluster.supervisor import ClusterView

class resizeGuiFonts(IPlugin):
    def attach_to_controller(self, controller):
	
        # Smaller font size for plot text  [default= 6.]
        TextVisual.default_font_size = 4.

        # Different method needed for Cluster & Similarity views
        # (note: not same font size units as plots)
        # We add a custom CSS style to the ClusterView.
        ClusterView._styles += """
        td {
            font-size: 6pt !important;
        }
        """
```

## The config
The catch/feature/bug with getting plugins applied is that along with putting the files in the correct directory ( `~/.phy/plugins` ...[_"there are many 'plugins' directories, but this one is mine"_]), they also need to be _'activated'_ by being named in the TemplateGUI plugins list.

```
# Put your plugins in ~/.phy/plugins/.

from phy import IPlugin

c = get_config()
c.Plugins.dirs = [r'~/.phy/plugins']
c.TemplateGUI.plugins = ['resizeGuiFonts']
```

For me, the most confusing part about this was that the plugin name needed to activate it, isn't necessarily (_often not_) the same as the .py file that contains the plugin! Its the name given to the class defined _within_ the plugin file.

So `c.Plugins.dirs = [r'~/.phy/plugins']` just makes all the .py plugin files in that directory 'known' to Phy, while `c.TemplateGUI.plugins = ['resizeGuiFonts']` is what actually makes the plugin active.

...I guess the idea is that you could put multiple plugins in one .py file and separately activate them, or maybe its just all evolutionary carryover from versions past.  Either way, hopefully this gets you back on track too.

ElKatz, 2020-07-02:
With our powers combined-- font size issue is solved!
Thanks for hanging in there Cyrill (and thanks of helping the community in general!), and thanks Thad!

rossant, 2020-07-03:
Great! and thank you @czuba  for the explanation, I updated the docs accordingly on the dev branch

closed 2020-07-03 by rossant