cortex-lab/phy #1011: GUI window messed up - can't change subpanels

Bionerdess, 2020-05-28:
I tried to open a data session today in the Phy GUI and the panels of the window are very messed up, don't have borders and the bar with dropdown menus disappeared, so I can't even delete or add subpanels. Because the borders are missing, I can't adjust the size of the windows. 
(see the attached image)

Everything worked fine on Monday. Maybe my computer updating over night changed something? Is anything known about a bug with Windows 10 version 1809?

Thank you!
[image]

rossant, 2020-05-31:
There is a bug with some of the Intel graphics drivers, could you update the drivers to the latest version?

Bionerdess, 2020-06-03:
Hi Cyrille,

Both the Intel and NVIDIA drivers are up to date and it is still not working.
Any other suggestions?

Thank you,
Anna-Lena

rossant, 2020-06-03:
What is the exact version of each driver? Do you have a way to switch to a different graphics driver when opening an application like phy?

Bionerdess, 2020-06-03:
Hi Cyrille,

This is the version of Intel driver: 26.20.100.7263 
It says the driver date is Sept 25 2019, which is weird, because it updated last week and I manually updated it again on Monday.

The NVIDIA driver is version 23.21.13.9125 (driver date March 16 2018; also updated on Monday).

I wouldn’t know how to switch to a different graphics driver when opening and application. Is there a command I can use before starting phy?

Thanks,
Anna-Lena

rossant, 2020-06-03:
The Intel driver doesn't seem to be the latest, see https://github.com/cortex-lab/phy/issues/957#issuecomment-609498355 (the correct one should be more recent than 26.20.100.7810)

Bionerdess, 2020-06-03:
> The Intel driver doesn't seem to be the latest, see #957 (comment) <https://github.com/cortex-lab/phy/issues/957#issuecomment-609498355> (the correct one should be more recent than 26.20.100.7810)

It told me that the best/latest driver for this computer is already installed. So I went to the Intel website and downloaded the 16.20.100.7870 driver (7810 does not exist on there). I tried installing it, but it gave me an error and told me it is not valid for this computer. 

I will also contact our IT person and see if he has some input.

Best,
Anna-Lena

hfcel, 2020-07-30:
Almost the same problem. I just updated both intel and nvidia graphic drivers as suggested by rossant, but nothing changed.

Intel(R) UHD Graphics 630
driver version: 26.20.100.7324

NVIDIA GeForce GTX 1650
driver version: 26.21.14.4343

hfcel, 2020-08-10:
> Almost the same problem. I just updated both intel and nvidia graphic drivers as suggested by rossant, but nothing changed.
> 
> Intel(R) UHD Graphics 630
> driver version: 26.20.100.7324
> 
> NVIDIA GeForce GTX 1650
> driver version: 26.21.14.4343

> Almost the same problem. I just updated both intel and nvidia graphic drivers as suggested by rossant, but nothing changed.
> 
> Intel(R) UHD Graphics 630
> driver version: 26.20.100.7324
> 
> NVIDIA GeForce GTX 1650
> driver version: 26.21.14.4343

I updated intel graphic driver by auto search of windows 11 days ago, and it didn`t work. Today I find it is not the latest version, and now the latest version works well.