cortex-lab/phy #1012: problem with dev install

nsteinme, 2020-06-02:
I just did a fresh install into a new conda env, specifically I created an empty env, cloned this repo, ran
pip install -r requirements.txt
pip install -r requirements-dev.txt
pip install -e .

I get error: 
ImportError: cannot import name 'SpikeSelector' from 'phylib.io.array' (C:\Users\nicks\Anaconda3\envs\phy2dev_20200602\lib\site-packages\phylib\io\array.py)

Indeed in the array.py at that location there is no SpikeSelector, though I see that there is in the phylib repo latest version. I'm not sure the correct step to fix it and I suspect I am about to enter conda hell if I get it wrong: can I just clone the latest phylib into that site-packages folder? do I need to pip install it with some additional flag like --upgrade or something? can I point the environment to a cloned version of the latest phylib elsewhere? please advise! (and fix/update install instructions, or let me know what I got wrong)

rossant, 2020-06-02:
I think a step is missing in the dev instructions, at the end:
pip install phylib --pre --upgrade

nsteinme, 2020-06-02:
Thanks for the quick reply but when I run that it shows:

Requirement already up-to-date: phylib in
c:\users\nicks\anaconda3\envs\phy2dev_20200602\lib\site-packages (2.2)

(among other things) and I get the same error.

rossant, 2020-06-02:
pip uninstall phylib
Then the command I gave you earlier
Does that work?
The problem is that requirements.txt installs the latest stable version of phylib, whereas phy currently requires the more recent beta version

nsteinme, 2020-06-02:
Still no - it again downloads 2.2 and I get the same error.

(phy2dev_20200602) D:\shiftTemp>pip install phylib --pre --upgrade
--no-cache-dir
Collecting phylib
  Downloading phylib-2.2-py2.py3-none-any.whl (89 kB)
     |████████████████████████████████| 89 kB 5.8 MB/s

rossant, 2020-06-02:
wait, do the same command but replace phylib with phy?

nsteinme, 2020-06-02:
It still installs phylib 2.2:

(phy2dev_20200602) D:\shiftTemp>pip install phy --pre --upgrade
--no-cache-dir
Requirement already up-to-date: phy in c:\users\nicks\dropbox\code\phy2\phy
(2.0b2.dev0)
Collecting phylib
  Downloading phylib-2.2-py2.py3-none-any.whl (89 kB)
     |████████████████████████████████| 89 kB 5.8 MB/s

Can I just drop latest phylib code into site-packages or some other
workaround?

rossant, 2020-06-02:
sorry for the confusion but phy is the one that should be upgraded to the latest beta version
stable phylib should be fine
if it still doesn't work just uninstall phy and phylib, clone both repos somewhere, and do python setup.py develop in each.

nsteinme, 2020-06-02:
Excellent, that worked (setup.py on both), thank you! I'm not 100% sure
you're right about phy not phylib needing upgrade: when phylib was
identifying as 2.2 the io.array file did not have the SpikeSelector. Now it
identifies as 2.3a0 and does have that. But entirely possible I've
misunderstood something.

rossant, 2020-06-08:
I fixed the dev install instructions accordingly

closed 2020-06-08 by rossant