cortex-lab/phy #1013: suggestion: freeze column labels on cluster and similarity view

nsteinme, 2020-06-02:
Where "freeze" means like in excel, so that no matter where you scroll down on the list, you can still see the column headers. It could be an option, but I think ought to be the default.

rossant, 2020-06-03:
I agree, I had tried to do that already and failed because of how the cluster view is currently implemented. This will probably require #951

czuba, 2020-07-13:
I managed to get this header fix implemented in this submitted [pull request](https://github.com/cortex-lab/phy/pull/1019/commits/3dae7e4881d31ee4fc143282137ad94af5f25133).

The catch was a high-dpi scaling that was being used when setting up the clusterView window (`transform` & `overflow`) interfered with the HTML sticky header implementation.
(...lots of google results about overflow being an issue, transformation issue was a brute force discovery)

Since the high-dpi scaling seemed to be producing [other side-effects](https://github.com/cortex-lab/phy/issues/1009), axing them seemed like a worthwhile trade-off.

nsteinme, 2020-08-10:
@rossant does this PR from czuba work? This issue is really painful - would be fantastic to have a solution merged.

rossant, 2020-08-31:
the pull request has been merged -- sorry for the delay!

closed 2020-08-31 by rossant