cortex-lab/phy #1015: template_ind

holmescharles, 2020-06-15:
I am using Spyking Circus to export sorted data to phy in a sparse format. I have data from 32 channels, but am excluding around half of them (the excluded channels are for all intents and purposes for this quesiton random).

Based on the website description, I would expect `template_ind.py` to represent which channels the templates are found on.

> templates_ind.npy - [nTemplates, nTempChannels] double matrix specifying the channels on which each template is defined. In the case of Kilosort templates_ind is just the integers from 0 to nChannels-1, since templates are defined on all channels.

But instead, it seems to be the channel's "index" in a list of unique channels. So rather than having values that correspond to the good channels (e.g., [1, 3, 5, 6, ...]) They are [-1, 0, 1, ...., n_good_channels-1]). Is this right?

yger, 2020-06-16:
if templates are saved in the "sparse" mode, then template_ind.npy is a matrix of size nb_templates x n_good_channels (where n_good_channels is the maximum of non zero channels across all the templates). 
Every line of this matrix should start with the indices of the non zeros channels for the given template, and then ends with -1. You should not have any line starting by -1

holmescharles, 2020-06-16:
Sorry, I wrote that in a bit of a hurry...., the values never start with a -1, but I put -1 in that list because they *CAN* be -1 at the end (indicating that its not a channel).
E.g., template_ind elements can be [0, -1, -1, -1] or [1, 3, -1, -1]. But you'll never see [-1, 0, -1, -1].

But I think I understand now --- the positive values correspond to elements in channel_map.npy, right?

yger, 2020-06-17:
Yes, and the -1 at the end of the row are only there because since not all templates are defined on the same amount of channel, we need to fill the values for those with less channels. But the first values are indeed the elements in channel_map.npy