cortex-lab/phy #1016: Crash GUI when clicking on item in ClusterView list while in Debug mode (PyCharm)

cboulay, 2020-06-17:
While running `template_gui` in Debug mode in PyCharm, clicking on an item in the Cluster View list is very likely to crash the app with the following error:

```
Traceback (most recent call last):
  File "d:\tools\neurophys\phy2\phy\phy\cluster\views\base.py", line 187, in finished
    @worker.signals.finished.connect
NameError: free variable 'self' referenced before assignment in enclosing scope
```

This usually happens on the first click after doing something else in the GUI. Because the error is triggered inside a qt event loop, I can't debug it very easily and I'm afraid learning the entire app design is beyond the time I have available.

The bug is not blocking me, it's just inconvenient while I'm trying to learn the correct format for the required data files. I intend to run the app from console when I'm actually using it in production, and so far I haven't been able to crash it in that mode. So this is just an FYI, I guess.

Cheers!

rossant, 2020-06-18:
This looks like a weird multithread Qt/Python interaction. I pushed a temporary workaround but I'm not sure that will really fix it.

cboulay, 2020-06-24:
I just got around to testing this. The workaround prevents the app from crashing, thanks!

Now, some of the GUI elements (waveform-related plots like ISI histogram) become unresponsive, and when I eventually close the window I get `QWaitCondition: Destroyed while threads are still waiting`.

Anyway, I've managed to figure out the data format and I was able to process a dataset so I'm satisfied. I doubt this bug is worth solving so I'll go ahead and close the issue. I'm curious to know what your dev environment is though, in case I do ever find myself wanting to modify/contribute.

Cheers.

rossant, 2020-06-25:
Thanks for the update.
I develop on Ubuntu 20.04, using Microsoft VSCode, and a terminal where I launch commands with `py.test` or `phy`