cortex-lab/phy #1018: GLError - cluster selected, nothing visualized in any "view" windows

Bionerdess, 2020-07-02:
Hi, 

I just installed phy on a new computer according to the guidelines. The cluster list appears, but all other visualization windows stay empty when a cluster is selected. (See attached image)
In the cmd window, the following error appears multiple times:

> [90m12:04:12.108 [D] base:673             Exception in initializetGL: GLError(
>         err = 1282,
>         description = b'invalid operation',
>         baseOperation = glClearColor,
>         cArguments = (0, 0, 0, 0)
> )[0m
> [90m12:04:12.109 [D] base:700             Exception in paintGL: GLError(
>         err = 1282,
>         description = b'invalid operation',
>         baseOperation = glClearColor,
>         cArguments = (0, 0, 0, 10)

Do I need to change something in the params.py file?

Thank you!
[image]

rossant, 2020-07-03:
* What is your system and graphics chipset?
* Can you update your graphics driver to the latest version by downloading them directly from the Intel/AMD/NVIDIA website?
* If you have a dual GPU (eg Intel and NVIDIA) do you have a way to switch to the non-Intel one?

Related:
* https://github.com/cortex-lab/phy/issues/899
* https://github.com/cortex-lab/phy/issues/987

neurongeek, 2020-07-03:
Same issue here. @Bionerdess did you get phy to work yet? I have the same issue :(
[image]

Bionerdess, 2020-07-06:
Ok it works now with 
Intel driver 27.20.100.8336 (Intel i9-9900K 3.6GHz)
And NVIDIA GeForce RTX 2080 Ti version 27.21.14.5148

However our older machine has a 6th generation Intel and can therefore not be updated to the latest Intel driver version. I brought this up in Issue #1011 and haven't been able to fix it on that computer.

joramvanrheede, 2021-04-28:
I'm currently having this issue (exactly the same as @Bionerdess originally described, with the same --debug output) on a shared Windows 10 machine with 2 NVIDIA GeForce RTX3090's with the latest drivers (27.21.14.6611). It also has an 'ASPEED Graphics Family (WDDM)' listed under display adapters, which seems to be some kind of Windows onboard video driver for Intel server boards (also with the most up to date driver version), I don't know if that is likely to interfere with anything?

It is a shiny new machine and a completely clean install (I've successfully installed and run Phy on other mac and windows computers before so I don't think I'm doing anything wrong on the installation front).

Will update if we manage to resolve the problem.

joramvanrheede, 2021-05-05:
Just an update, I'm trying to figure out whether the fact that I am using Windows Remote Desktop to access this machine may be the problem, as in issues #945 & #737. However, the issue persists when using Chrome remote desktop to open phy. Unfortunately it is a shared server and I do not have physical access to it to test whether running it locally will resolve the issue.

joramvanrheede, 2021-05-05:
Phy is now working! The issue was indeed with Windows Remote Desktop using OpenGL 1.1. It took a surprisingly long amount of searching before I found out that actually, Windows Remote Desktop *can* run higher OpenGL versions and give the remote session access to the GPU, but it is disabled by default in a Group Policy.

Steps to solve the problem:
1) I installed OpenGL extensions viewer to see what OpenGL Version was being used, this was version 1.1 for me (when phy was broken) - link: http://www.realtech-vr.com/home/glview

2) Enable default hardware graphics adapter using the steps from the following link: https://community.esri.com/t5/implementing-arcgis-questions/enabling-gpu-rendering-on-windows-server-2016-windows-10-rdp/td-p/658522
Key info (quoted from link):

>     Open the Edit Group Policy tool from Control Panel or use the Windows Search dialog (Windows Key + R, then type in gpedit.msc)
    Browse to: Local Computer Policy\Computer Configuration\Administrative Templates\Windows Components\Remote Desktop Services\Remote Desktop Session Host\Remote Session Environment
    Then enable “Use the hardware default graphics adapter for all Remote Desktop Services sessions”

After a re-start and logging in again, OpenGL extensions viewer now shows OpenGL version 4.6 and phy displays are all working.

This solution may also fix issues #945 & #737