cortex-lab/phy #1020: Views not updating when cluster selected (running Phy on VMware virtual machine)

zuzapiwkowskazvonkine, 2020-07-06:
Hi, I have recently started using kilosort2 and phy for Neuropixels data, and there are some issues with display updates. I'm wondering if it could be due to the fact that I am doing all this remotely on a virtual machine (VMware). 

Specifically, the updating of the different views is quite random when I select a cluster. I also have the feeling it got worse after I moved some of the windows around to have the GUI the way I preferred: now upon selection of a cluster, most windows only update to the selected cluster when I close them and open them again from the View menu. And sometimes they do not update at all.

This is all for:
phy, version 2.0b1
Anaconda3 (64-bit)
Windows 10 Enterprise LTSC
NVIDIA GRID RTX8000-8Q GPU

...but since the display happens remotely, I guess it must be using something else? I also see VMware Horizon Indirect Display Driver in the device manager. I really don't know much about virtual machines at this point.

Did anybody also have issues running phy on a VM? Or are there some other debugging steps I should try? (btw --debug mode does not give any error messages)

Thank you,
Zuzanna

zuzapiwkowskazvonkine, 2020-07-06:
Ok, after playing around some more, I'm concluding that this updating issue is basically a performance issue and that it is mostly solved by making sure TraceView is really zoomed in to only show a very short piece of data. What I'm still wondering about is whether the performance on displaying this same data would be significantly better if I did not work on the VM. If anybody has insights into this please let me know!
Best,
Zuzanna

rossant, 2020-07-06:
The VM might be a cause of this problem. Do you know if it uses GPU acceleration?
You should be able to disable the automatic update of the trace view upon cluster selection by clicking on the little circling arrows at the top right of the trace view

zuzapiwkowskazvonkine, 2020-07-06:
I also see the VMware SVGA 3D driver in my display adapters... does this stand for shared virtual graphics acceleration? No idea how to check otherwise but I can ask some experts. Thanks for the tip about the circling arrows, I keep forgetting about this, it also helps!

zuzapiwkowskazvonkine, 2020-07-06:
Everything seems to also work much faster now (ie after regular work hours) all of a sudden, so possibly it is a network traffic issue mostly (ie indirectly related to the VM). Thanks for your insights, I think the issue can probably be closed.
All the best,
Zuzanna

annclem, 2020-09-30:
I am also having trouble with views freezing and not updating when I move between clusters though I am working on the machine locally. I just installed everything two days ago so I don't know if some settings need to be adjusted?

using:
Windows 10 Pro
NVIDIA Quadro RTX 6000
Anaconda 3
phy2.0b1

annclem, 2020-09-30:
oh, i just did this and it works much better: phy template-gui params.py --clear-state

maybe that solved it...