cortex-lab/phy #1021: PyQt5.QtWebEngineWidgets missing from install

rglanz, 2020-07-07:
I installed using the developer instructions and got this error.

`ModuleNotFoundError: No module named 'PyQt5.QtWebEngineWidgets'`

Looks like it got moved to a separate package. Running `pip install PyQtWebEngine` fixed the issue.

rossant, 2020-07-07:
Yes this is a known issue: https://github.com/cortex-lab/phy#dealing-with-the-error-modulenotfounderror-no-module-named-pyqt5qtwebenginewidget
The problem is that it looks like this command should only run on particular environments (combination of OS and Python distribution), and not others otherwise it fails, and it's unclear to me which!

NixPlayer, 2022-09-01:
just run in a terminal:
pip install PyQtWebEngine