cortex-lab/phy #1023: View windows frozen

njmiska, 2020-07-22:
Hi,

A few of us in the IBL have been encountering the issue that after adding/deleting any view windows, they all will freeze at that point and fail to update upon clicking additional clusters.  The only way to resolve it seems to be to quit and reopen phy.

We are using phy_launcher as part of iblapps.  The issue can be replicated (on my machine) by opening a dataset in phy and clicking the x in the upper right hand corner of one of the view windows.  All of the views will be subsequently 'frozen'.

Nate

rossant, 2020-07-22:
Hi, thanks for the bug report. I just pushed a tentative hot fix on the master branch, could you update and see if that fixed it?

njmiska, 2020-07-23:
Hey Cyrille,

Yeah that fixed it, thanks!

Cheers,
Nate