cortex-lab/phy #1026: Can't launch after install

olinesn, 2020-08-10:
Hello,

I'm having difficulty getting phy to read the output from Kilosort2. I can see the params.py file in the directory that I've cd'd to. However, when I follow the steps in the README and get to step 3, I get stuck at `phy template-gui params.py`

Error: Invalid value for 'PARAMS_PATH': Path 'params.py' does not exist.

[image]

Is there something that I could be doing differently here?

Thanks!

rossant, 2020-08-10:
It looks like the `cd H:\...` command didn't work (the prompt is still in the `C:\Users\duo41` directory).

Try typing `H:`, then press Enter, then type again the  ̀cd H:\...` command followed by the `phy template-gui params.py` command

olinesn, 2020-08-10:
> It looks like the `cd H:\...` command didn't work (the prompt is still in the `C:\Users\duo41` directory).
> 
> Try typing `H:`, then press Enter, then type again the `cd H:\...` command followed by the `phy template-gui params.py` command

Thanks for your quick reply. Sorry that was an unrelated type, getting the same error thrown even if I execute the cd correctly. However I'm now looking at the params.py output from my KS2 and seeing that the file is only 1KB. My intuition is then that the export didn't work correctly, I'll attack from that angle.
[image]

rossant, 2020-08-10:
You're still not in the right directory: now that the prompt shows `H:\>`, type the full `cd H:\...` command. Then type the `phy template-gui params.py` command.
The params.py is always a small 1KB text file.

olinesn, 2020-08-10:
Thanks for your patience, I don't have a background working in a terminal. Unfortunately it looks like the issue persists, in spite of arriving at the correct directory. Is there a common user error for when the params.py is in the folder, but not recognized by Phy2?

[image]

rossant, 2020-08-10:
This is strange. Could you post (1) a screenshot of the directory showing all the files in it, and (2) the params.py file itself?