cortex-lab/phy #1027: Custom view to plot amplitude on one site versus another

audrocks17, 2020-08-12:
Hello Phy team,

My lab mates and I have been working to design a custom view or plugin for phy. The software that our lab has traditionally used to manually isolate clusters has a feature where the max amplitude of a spike on one channel of the probe can be plotted on the x axis versus its max amplitude on any other selected channel on the y axis. It would be great if we could have a similar view in phy so that we can plot and modify cluster bounds of kilosort isolated clusters in a similar way. I have tried to approach this by modifying the example plugins in the documentation. The goal would be something like a hybrid between the native feature view and amplitude view. For example, if it were possible in feature view to plot amplitudes, or if in amplitude view one could plot the raw amplitude on the best channel on the y axis versus amplitude on the second best channel on the x axis (ideally it would be able to plot from any channel). I am relatively new to coding and have sort of hit a wall with this and thought I might ask for advice. I have been trying to modify the example plugin under “customizing the feature view”. Would it be possible to use something like this to change the axis dimensions to be raw spike amplitudes? If so do you have a suggestion of how I might approach this?

# import from plugins/feature_view_custom_grid.py
"""Show how to customize the subplot grid specifiction in the feature view."""

import re
from phy import IPlugin, connect
from phy.cluster.views import FeatureView

def my_grid():
    """In the grid specification, 0 corresponds to the best channel, 1
    to the second best, and so on. A, B, C refer to the PC components."""
    s = """
    0A,1A 1A,2A 2A,0A
    0B,1B 1B,2B 2B,0B
    0C,1C 1C,2C 2C,0C
    """.strip()
    return [[_ for _ in re.split(' +', line.strip())] for line in s.splitlines()]

class ExampleCustomFeatureViewPlugin(IPlugin):
    def attach_to_controller(self, controller):
        @connect
        def on_view_attached(view, gui):
            if isinstance(view, FeatureView):
                # We change the specification of the subplots here.
               view.set_grid_dim(my_grid()) 

Thank you so much in advance!

audrocks17, 2020-08-31:
I have made some progress on this since posting, making use of the umap view plugin example as a base. This plugin will successfully plot amplitudes on the best channel. I now would like to extend it so that one can plot amplitude on channels neighboring the best channel for a cluster, ideally in grid format. Do you have a suggestion of how I might modify this to do so?

import numpy as np
from phy import IPlugin, Bunch
from phy.cluster.views import ScatterView

class WaveformUMAPView(ScatterView):
    """Every view corresponds to a unique view class, so we need to subclass ScatterView."""
    pass

class umap_view(IPlugin):
    def attach_to_controller(self, controller):
        def coords(cluster_ids):
            spike_ids = controller.selector.select_spikes(cluster_ids)
	        # We get the cluster ids corresponding to the chosen spikes.
            spike_clusters = controller.supervisor.clustering.spike_clusters[spike_ids]

            data1 = controller.get_spike_raw_amplitudes(spike_ids, channel_id=controller.get_best_channels(spike_clusters[0])[0])
            data2 = controller.get_spike_raw_amplitudes(spike_ids, channel_id=controller.get_best_channels(spike_clusters[0])[1])
            print(controller.get_best_channels(spike_clusters[0])[0], controller.get_best_channels(spike_clusters[0])[1])

            pos = np.ones((len(data1), 2))
            pos[:,0] = data1
            pos[:,1] = data2

            return Bunch(pos=pos, spike_ids=spike_ids, spike_clusters=spike_clusters)

        def create_view():
            """Create and return a histogram view."""
            return WaveformUMAPView(coords=controller.context.cache(coords))

        # Maps a view name to a function that returns a view
        # when called with no argument.
        controller.view_creator['WaveformUMAPView'] = create_view

rossant, 2020-08-31:
Sorry for the delay getting back to you. For your use case you might have to derive your WaveformUMAPView from FeatureView instead, which is tricker as it has subplots.

The idea is that you pass to the constructor of the class a function which maps cluster ids to data points. You may achieve what you want just by defining a proper `features` function that returns the spike amplitudes on several channels. Otherwise you may have to override some internal methods here: https://github.com/cortex-lab/phy/blob/master/phy/cluster/views/feature.py#L158

audrocks17, 2020-08-31:
No problem, thank you for responding! And thank you for this suggestion, I will give this a shot...

audrocks17, 2020-09-21:
Hello again! Thank you for the suggestion regarding working from FeatureView. I have been able to create a version of FeatureView that plots data based on _get_spike_raw_amplitudes, and plots different channels relative to one another. I have a few questions about how I could maybe specify the channels in a different way. First, _get_spike_raw_amplitudes gets the maximum amplitude of the raw waveform on the best channel, and I would like to get raw amplitudes on neighboring channels, higher up and lower down on the probe. For example, if the best channel for a cluster is channel 10, I would like the grid to be set up as: 

raw_amp 8, raw_amp 9
raw_amp 9, raw_amp 10
raw_amp 10, raw_amp 11
raw_amp 11, raw_amp 12
raw_amp 12, raw_amp 13

I have tried setting this up by doing:

    -2A,-1A
    -1A,0A
    0A,1A
    1A,2A
    2A,3A

But it looks a little strange on the plots. Is this doing what I am hoping for? Or is there a better way to go about this? Thank you in advance!