cortex-lab/phy #1031: GUI crashes with segmentation fault after clicking on GUI

afergu, 2020-09-14:
I open phy successfully. I click on one unit and all windows update to show the waveform. Upon clicking anywhere on the GUI, it crashes. Occasionally if I click a unit and then wait a while, it will also crash. This happens on any file I try. 

- macOS 10.15.6
- phy is on `master` (currently at `2.0b2.dev-git-v2.0a1-201-g38ff8e29`); before I checked out the dev build I was having this same issue on the most recently released version (which is why I checked out the dev build in the first place!)

<details><summary> Full stack trace </summary>

```
Process:               python3.7 [25048]
Path:                  /opt/anaconda3/*/python
Identifier:            python3.7
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        bash [24631]
Responsible:           Terminal [480]
User ID:               501

Date/Time:             2020-09-14 10:46:07.515 -0400
OS Version:            Mac OS X 10.15.6 (19G2021)
Report Version:        12
Bridge OS Version:     4.6 (17P6610)
Anonymous UUID:        3C8EE0FC-52A5-2DD6-F822-278DBFD8DE82

Sleep/Wake UUID:       7F9D6E23-1A11-4CCB-BB3A-C48D43DBCE8F

Time Awake Since Boot: 34000 seconds
Time Since Wake:       7900 seconds

System Integrity Protection: enabled

Crashed Thread:        39  Thread (pooled)

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [25048]

VM Regions Near 0x18:
--> 
    __TEXT                 000000010bc10000-000000010becf000 [ 2812K] r-x/rwx SM=COW  /opt/anaconda3/*/*.7

Thread 0:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   QtCore                        	0x000000011942019c QHashData::nextNode(QHashData::Node*) + 28
1   QtCore                        	0x00000001195d783a buildMetaObject(QMetaObjectBuilderPrivate*, char*, int, bool) + 5466
2   QtCore                        	0x00000001195d62d1 QMetaObjectBuilder::toMetaObject() const + 65
3   QtCore.abi3.so                	0x00000001192104ce PyQtSlotProxy::PyQtSlotProxy(_object*, QObject*, Chimera::Signature const*, bool) + 398
4   QtCore.abi3.so                	0x000000011920bc11 get_receiver_slot_signature(_object*, QObject*, Chimera::Signature const*, bool, QObject**, QByteArray&, bool, int) + 289
5   QtCore.abi3.so                	0x000000011920bfb0 pyqtBoundSignal_connect(_object*, _object*, _object*) + 272
6   python                        	0x000000010bc3c2fb _PyMethodDef_RawFastCallKeywords + 395
7   python                        	0x000000010bd6cb7c call_function + 364
8   python                        	0x000000010bd64ebb _PyEval_EvalFrameDefault + 22187
9   python                        	0x000000010bd5e157 _PyEval_EvalCodeWithName + 551
10  python                        	0x000000010bc3a842 _PyFunction_FastCallDict + 450
11  python                        	0x000000010bc3eaaa method_call + 122
12  python                        	0x000000010bc3c83f PyObject_Call + 127
13  python                        	0x000000010be2e7c6 partial_call + 262
14  python                        	0x000000010bc3c83f PyObject_Call + 127
15  python                        	0x000000010bd65066 _PyEval_EvalFrameDefault + 22614
16  python                        	0x000000010bd5e157 _PyEval_EvalCodeWithName + 551
17  python                        	0x000000010bc3a842 _PyFunction_FastCallDict + 450
18  python                        	0x000000010bc3eaaa method_call + 122
19  python                        	0x000000010bc3c83f PyObject_Call + 127
20  python                        	0x000000010bd65066 _PyEval_EvalFrameDefault + 22614
21  python                        	0x000000010bc3a770 _PyFunction_FastCallDict + 240
22  python                        	0x000000010bc3eaaa method_call + 122
23  python                        	0x000000010bc3c83f PyObject_Call + 127
24  python                        	0x000000010bd65066 _PyEval_EvalFrameDefault + 22614
[... 1282 lines cut ...]
10:56:01.937 [D] base:334             Set state for ISIView.
10:56:01.997 [D] gui:718              Add view FeatureView to GUI.
10:56:02.000 [D] base:334             Set state for FeatureView.
10:56:02.024 [D] gui:718              Add view AmplitudeView to GUI.
10:56:02.026 [D] base:334             Set state for AmplitudeView.
10:56:02.048 [D] gui:718              Add view FiringRateView to GUI.
10:56:02.051 [D] base:334             Set state for FiringRateView.
10:56:02.187 [D] gui:718              Add view TraceView to GUI.
10:56:02.190 [D] base:334             Set state for TraceView.
10:56:02.190 [D] base:490             Set color scheme to depth.
10:56:02.209 [D] base:1623            Adding default color schemes to TraceView.
10:56:02.270 [D] gui:718              Add view ProbeView to GUI.
10:56:02.272 [D] base:334             Set state for ProbeView.
10:56:02.288 [D] gui:718              Add view TemplateFeatureView to GUI.
10:56:02.290 [D] base:334             Set state for TemplateFeatureView.
10:56:02.310 [D] gui:718              Add view ClusterScatterView to GUI.
10:56:02.312 [D] base:334             Set state for ClusterScatterView.
10:56:02.313 [D] base:490             Set color scheme to firing_rate.
10:56:02.313 [D] base:1623            Adding default color schemes to ClusterScatterView.
10:56:02.455 [D] gui:519              Load the geometry state.
10:56:08.089 [D] supervisor:813       Clusters selected: [33] (26)
10:56:09.165 [D] supervisor:835       Similar clusters selected: [23] (104)
Segmentation fault: 11
```
</details>

afergu, 2020-09-28:
It turns out I was trying to look at ks1 sorted data in phy2. This causes the GUI to crash. Is that always the case? If so, can you add it to the README so others don't deal with this too.

rossant, 2020-09-28:
@afergu thanks you for the information. No, it is not expected, and phy2 ought to work with ks1 data. If you're able to send me your dataset I can try opening it on my Linux machine as well as a Macbook Pro.

afergu, 2020-09-29:
I can send you data. It's large and I'd rather not have it on this issue publicly. Can I pm you a dropbox link?

rossant, 2020-09-30:
Thanks, I was able to open your dataset on my Linux machine, and I couldn't reproduce the issue. I'll try on macOS within the next few days.

But looking at the stack trace it seems that the problem comes from the Web-based cluster view which has caused many issues. This might only be fixable via https://github.com/cortex-lab/phy/issues/951 which will only happen in a few months...

afergu, 2020-10-02:
When I used ks2 to sort my data, it reported many 'bad channels' (~40/64channels) but it opened successfully without crashing the GUI. 

However I changed 

```
ops.minfr_goodchannels = 0
```
to stop ks2 from removing those channels. 

Now when I try to load this dataset in phy, it crashes in the same way as for ks1 data. I will also pm you this data set

rossant, 2020-10-04:
I tried both datasets on my macOS 10.15.6 (macbook pro 2015) on master and was unable to reproduce the issue. I'm out of suggestions unfortunately :(

afergu, 2020-10-07:
Thanks so much for taking a look! I'll try it on another machine and see what happens.

closed 2021-09-27 by afergu