cortex-lab/phy #1032: Waveform View won't update

macayladonegan, 2020-09-14:
Hi,

I recently switched from sorting with Kilosort to Kilosort2, and now when I visualize my data in phy2 my waveform view won't update/let me toggle between the different waveform views.  It seems to be stuck on the average waveform for the first cluster I select.  Any advice?

Best,
Macayla

rossant, 2020-09-18:
What is your operating system? Could you send me your dataset so that I check whether it works on my computer?

macayladonegan, 2020-09-21:
Windows 10, here's a drive link

macayladonegan, 2020-09-21:
oops, here it is:
https://drive.google.com/drive/folders/1joQ64chhRVuWpPgcrFmgE_SCQqlAsaZQ?usp=sharing

annclem, 2020-09-29:
I think I may be encountering the same problem. I see many spikes in all other windows but only a single one in WaveformView. Also using Windows 10. Any ideas on what the issue might be?

MHRosenberg, 2020-11-05:
Was this ever resolved? I'm having a similar issue in ubuntu 16.04. I have plenty of RAM but after assigning a few clusters the waveform view ceases to update. I can provide my dataset upon request.

blaje42, 2020-11-12:
I am also having a similar issue on Windows 10 with 64 GB of RAM. Waveforms cease to update and none of the toggle hotkeys for waveform view are working for me.

rossant, 2020-11-13:
This only affects the waveform view? The other views are still updated when cluster selection changes? Hotkeys of other views continue to work?

blaje42, 2020-11-13:
No, I am also having this issue with the correlogram view more so than the other views. I have not tried all the views, but I have had no trouble with hotkeys for TraceView or when rating clusters.

grahamfindlay, 2021-01-15:
I also had this issue using output from Kilosort 2.5, Windows 10, latest version of phy2. It appears to be caused by closing the ProbeView. 

More observations:
- Once ProbeView is closed, the following plots continue to auto-update: ClusterScatterView, SimilarityView, TraceView (when forced w/ e.g. Alt+PgDown). No other plots auto-update, though scaling, changing ISI bin sizes, etc. still works. 
- Inspecting `state.json` in the Template-GUI directory of your user folder shows that auto-update is still set to `true` for plots that do not auto-update. 
- Closing and re-opening the WaveformView will cause the new WaveformView to auto-update, but keyboard shortcuts for that view no longer work. For example, trying "W" to switch waveform type gives the error: `QAction::eventFilter: Ambiguous shortcut overload: W`. Furthermore, other frozen views like ISIView will not auto-update until they are also closed and re-opened. 

So for now, my solution is just to minimize the ProbeView as much as possible, or undock and hide it, but never close it.

ClaudiaJou, 2021-01-25:
Hi,
Having the same issue here also using the output from Kilosort 2.5 in the latest version of phy2, on a Windows 10 PC. Other views also freeze, not only WaveformView. Keeping ProbeView open only worked for a few minutes.

blaje42, 2021-01-29:
Keeping ProbeView open also did not help for me, unfortunately.