cortex-lab/phy #1035: ValueError: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size. [0m

miguelconcham, 2020-09-23:
I can not load the Kilosort2 output into the GUI. I pasted the error below. Should I try to run Kilosort with a different configuration so the output is somehow smaller? I have not had this problem before, using the same type of recordings, and similar data sizes.
 
Best

Miguel

c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\numpy\core\memmap.py:245: RuntimeWarning: overflow encountered in int_scalars
  size *= k
c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\numpy\core\memmap.py:247: RuntimeWarning: overflow encountered in int_scalars
  bytes = int(offset + size*_dbytes)
[31m10:46:42.500 [E] __init__:62          An error has occurred (ValueError): array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.
Traceback (most recent call last):
  File "c:\users\neuropixels\.conda\envs\phy2\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\neuropixels\.conda\envs\phy2\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phy\apps\template\gui.py", line 198, in template_gui
    controller = TemplateController(model=load_model(params_path), dir_path=dir_path, **kwargs)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phylib\io\model.py", line 1178, in load_model
    return TemplateModel(**get_template_params(params_path))
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phylib\io\model.py", line 297, in __init__
    self._load_data()
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phylib\io\model.py", line 389, in _load_data
    self.traces = self._load_traces(self.channel_mapping)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phylib\io\model.py", line 523, in _load_traces
    for path in paths]
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phylib\io\model.py", line 523, in <listcomp>
    for path in paths]
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phylib\io\model.py", line 176, in load_raw_data
    return _dat_to_traces(path, n_channels=n_channels_dat, dtype=dtype, offset=offset, order=order)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\phylib\io\model.py", line 149, in _dat_to_traces
    str(dat_path), dtype=dtype, shape=(n_samples, n_channels), offset=offset, order=order)
  File "c:\users\neuropixels\.conda\envs\phy2\lib\site-packages\numpy\core\memmap.py", line 267, in __new__
    offset=array_offset, order=order)
ValueError: array is too big; `arr.size * arr.dtype.itemsize` is larger than the maximum possible size.
[0m

rossant, 2020-09-23:
Can you double check that you have a 64-bit installation of Python? You can also do

`python -c 'import sys;print("%x" % sys.maxsize, sys.maxsize > 2**32)'`

miguelconcham, 2020-09-23:
Great! You were right.

It was different computer with a  32-bit version of python.

thank you very much!

rubayetalamnse, 2023-09-26:
mine is running in 64-bit machine still facing problem!