cortex-lab/phy #1039: Error with "phy template-gui params.py"

ensorpalacios, 2020-10-14:
Hi,

I'm on Centos 7 (tried also on Ubunto). I created the conda env for phy2 using the instructions:

conda create -n phy2 python=3.7 pip numpy matplotlib scipy scikit-learn h5py pyqt cython pillow -y
conda activate phy2
pip install phy --pre --upgrade

Then I cd in the directory where params.py is, type "phy template-gui params.py --debug" in the phy2 env, but I get this error:

17:25:34.084 [D] __init__:68          Start capturing exceptions.
17:25:34.122 [E] __init__:62          An error has occurred (PermissionError): [Errno 13] Permission denied: '/mnt/bunaken/Ensor/npx/EP_acute_1_g0/phy.log'
Traceback (most recent call last):
  File "/home/yq18021/miniconda3/envs/phy2/bin/phy", line 8, in <module>
    sys.exit(phycli())
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/gui.py", line 195, in template_gui
    _add_log_file(dir_path / 'phy.log')
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/phylib/__init__.py", line 65, in _add_log_file
    handler = logging.FileHandler(filename)
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/logging/__init__.py", line 1087, in __init__
    StreamHandler.__init__(self, self._open())
  File "/home/yq18021/miniconda3/envs/phy2/lib/python3.7/logging/__init__.py", line 1116, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
PermissionError: [Errno 13] Permission denied: '/mnt/bunaken/Ensor/npx/EP_acute_1_g0/phy.log'

QWidget: Must construct a QApplication before a QWidget
Aborted (core dumped)

Because I see permission denied, I tried to open the gui with "_sudo ~/miniconda3/envs/phy2/bin/phy template-gui params.py_" but no chance:

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
17:28:43.246 [W] model:545            Unreferenced clusters found in templates (generally not a problem)
17:28:43.319 [W] model:567            Unreferenced clusters found in spike_clusters (generally not a problem)
17:28:44.330 [I] state:180            Copied /home/yq18021/miniconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/template/static/state.json to /root/.phy/TemplateGUI/state.json.
[5919:5919:1014/172844.680552:ERROR:zygote_host_impl_linux.cc(89)] Running as root without --no-sandbox is not supported. See https://crbug.com/638180.

Any idea what I could be doing wrong?

Thanks

closed 2020-11-25 by ensorpalacios

seokhee-kang, 2023-10-02:
Hi I have this same error, would it be possible for you to share your solution to this?

ensorpalacios, 2023-10-05:
Hi, sorry I can't exactly remember how I solved this but from the logs it seams a permission issue. I think changing the permissions of the folders with chmod should work.

LordKelv, 2024-03-28:
> Hi, sorry I can't exactly remember how I solved this but from the logs it seams a permission issue. I think changing the permissions of the folders with chmod should work.

Heya!
Would you be able to expand on this?  I am having the same issue as you but don't know how to adjust permissions.
Thanks,
Kelvin