cortex-lab/phy #1041: Feature view changes after re opening a session

KaiserArndt2, 2020-10-26:
Is there a known solution to the issue I am where all of the dots in the feature view are stacked on top of one another and don't spread out until I re open phy2?

[image]

[image]

rossant, 2020-10-27:
it looks the channels are not properly selected (the channels are not the same between the 2 pictures)
there is an option in the menu to automatically show the best channels when you select another cluster
you can also manually change the channels shown in the feature view by ctrl+click on the relevant channels in the waveform view

KaiserArndt2, 2020-10-27:
Great, thank you so much!