cortex-lab/phy #1042: Adding lines marking stimuli blocks to AmplitudeView

lshaheen, 2020-10-28:
Hi,

I just tried out V2, it's looking great, lots of cool updates. For version 1 we have a plugin that plots the feature template projections as a function of time. Now I see that's rolled into the AmplitudeView and that's great! Inside that old plugin is also some code that would draw lines marking the start of different stimulus sets recorded over the course of the experiment. It would be awesome to get that back but I'm not quite sure where to begin. I made a brief attempt at writing a plugin to modify AmplitudeView but didn't get very far. Is it possible to have a plugin that would add more visuals on top of everything like that? I think I know how to override on_select, but I'm not sure how to just add to it. I'm also not very clear on how to use the new base visual classes to get lines in the right places.

[Here's the plugin for phy V1](https://github.com/LBHB/phy-contrib/blob/master/phycontrib/LBHB_plugins/FeatureTemplateTimeView.py). We usually concatenate several open-ephys recordings and run them through Kilosort. Each recording are the responses to a different class of stimuli. blocksizes.npy lists the number of samples in each recording and the plugin just drew vertical lines to mark those times. Since there's also some time between recordings, I also had an option to use blockstarts.npy, which listed the start times of each recording in samples, to make blank space between spikes from each recording. But this turned out to be not super useful anyway, most important would just be to get blocksizes.npy working.

I'm happy to take another try at this, but if someone who knows the code a bit more could give some pointers that would be most appreciated.

lshaheen, 2020-11-17:
Hi @rossant, would you mind just pointing me in the right direction on this?

rossant, 2020-11-18:
Hi, I'm sorry about the delay! This would be a long-awaited feature, it will be done eventually but not sure when. In the meantime, others have done similar things, see eg this thread https://github.com/cortex-lab/phy/issues/839 . Perhaps you could reuse some of it.

The visual API in phy2 is not well documented and is not expected to last very long, so perhaps not worth investigating it too much. I've been working for a long time on a much more powerful and robust visualization system, but that's not going to be ready until some time next year.