cortex-lab/phy #1044: No correlogram?

Idavr, 2020-10-30:
Hi there!

I am going over some sessions in phy and for quite a few units the CorrelogramView is completely blank. I haven't found a potential reason for this if it is by design (some insufficient information from the unit, for example) or a bug? If due to issues with the neuron, are there any workarounds to get more information about the degree of potential violations of the refractory period?

rossant, 2020-11-13:
Can you check the bin size and window length of the correlogram view? Or try to change it from the menu bar?

Idavr, 2020-11-14:
Bin size is 1ms, window length is 20ms. No problems changing them, but also still an empty window regardless of what I change them to.

gcresp2, 2026-10-29:
Hello, I was wondering if you were able to figure out what was causing this bug? I am sorting a session and for some units (not all in this recording) when I sort out noise the correlogram view disapears for the unit. I closed the view for the correlgram window and then reopened it and it was still empty. Then i selected another unit and the correlgram view was totally fine for that one. I am using  
phy                     2.1.0rc1   
Python 3.12.12  
pip 
Local version and remote 
no plugins 
no error message
the affected unit looks like:  

[image]

rossant, 2026-10-29:
Thanks for the details. Your screenshot shows a 9 ms correlogram window with 1 ms bins. A window this narrow can be empty if there are no sufficiently close spike pairs. Could you try a **100 ms window**, keeping the bin size at 1 ms, and share the result?

If it remains empty, please save the current curation and share the matching `spike_times.npy` and `spike_clusters.npy`, the affected cluster ID (the correlogram labels appear to show **909**), and `sample_rate` from `params.py`. These will let us check the calculation directly.

gcresp2, 2026-10-30:
Thank you so much Rossant! Increasing the window to 100ms and resetting the cache for this session fixed it!! 
For this unit in particular i had to increase the window to 150 in order to visulaize the baseline, but having a window of 100 helps with the other neurons!! 

I have included screenshots: 

[image]

[image]