cortex-lab/phy #1045: Waveform view does not rescale when i toggle between waveform types

boydmeredith, 2020-11-02:
Hi - I find the waveform view to be very helpful and as I curate my data, I like to toggle between the waveforms, the average and the template. Unfortunately, the template and the waveform have very different scales, so if I'm looking at the waveforms at a reasonable scale and I switch to the templates, I will only see flat lines. On the other hand, if I switch from template to waveforms the waveforms are too large too see. The only solution I've found is to close the GUI and reopen it with the waveform type selected. I have tried: clicking  'reset scaling', add template_scaling = 20.0 to params.py, and installing the development version. Nothing seems to be working (although, I'm not completely sure that I configured my environment properly to make sure I'm using the dev version). 

Thanks!

rossant, 2020-11-03:
The scaling discrepancy between the different types of waveforms is a long-standing issue that the ongoing model refactoring should fix (available for testing in a few weeks hopefully)

boydmeredith, 2020-11-03:
good to know. thanks! does that mean the changes will be in kilosort or is that something that happens in phy?

rossant, 2020-11-13:
The fix will be in phy

epmgs97, 2021-05-20:
I still have the same issue and it's been half a year... Could it be a problem with the filter scale reajustment that Kilososrt does when it takes our data to preprocess it? Maybe our data is in a different scale than the one used by Kilosort to represent the filtered waveform, so then when Phy tries to painted it seems like no wave is been captured, even though Kilosort does identify the spike.

Does anyone knows the range of values Kilosort wants to be given the raw data in the .bin file so the filter could finally be well visuelized in Phy?

emilyasterjones, 2022-05-16:
@rossant was this fix ever implemented? I'm having the same issue (Phy v2.0b5 + most recent Github pull of KS3). The template_scaling fix  in params.py and 'reset scaling' continue to not fix the issue.