cortex-lab/phy #1052: ClusterView/clust_info.tsv columns disappear during curation

boydmeredith, 2020-11-11:
Sometimes as I am curating a session, I find that the amplitude, contamPct, and KSLabel columns of the ClusterView go blank and their corresponding entries in clust_info.tsv go away too. Is there a way to avoid this or have it refresh? I want to use the clust_info.tsv file downstream and it is strange to have to go hand edit it.

Thanks!

lkatsouri, 2020-11-27:
I have the same issue, has anyone got a solution?

MMMariel, 2021-06-08:
I had the same in one of my sessions. Any update?

czuba, 2021-06-15:
Same experience here. 

I suspect this occurs because those columns comprise _values that are only computed within Kilosort_, and don't get recomputed when clusters are merged in Phy. 

Instances where they _appear_ to be maintained after manual curation are only those where a single cluster was split, and the old —but not necessarily valid—values have been inherited from the original cluster.