cortex-lab/phy #1054: Selection of spikes in Waveform View

sv-bio, 2020-11-20:
Dear Cyrille,
Thank you for such a useful visualization tool. A short question regarding waveform view, that I could not find answer for in the forum: often I find that clusters have waveforms that were apparently mis-classified and are clearly different from the rest (peaks at different time, double peaks etc.; see figure as an example). How could I select these waveforms and move to a "garbage/noise cluster"?

[image]

Thanks,
Stan

rossant, 2020-11-24:
Unfortunately selection in the waveform view is not possible at the moment

sv-bio, 2020-11-26:
Thanks for the answer. Just as a side note, I found IMO handy implementation of this functionality in wave_clus Matlab toolbox.

closed 2020-11-26 by sv-bio