cortex-lab/phy #1055: Feature View does not show clusters

sv-bio, 2020-11-20:
Dear Cyrille,
May I ask another question: rather often I find that even if several clusters are selected, Feature view shows something strange (see figure). If I zoom in, temporal distribution is displayed but nothing else otherwise. The file pc_features.npy exists. What could be the reason for such behavior?

[image]

Sincerely,
Stan

rossant, 2020-11-24:
It might be a gain problem in the feature view, you can increase it by pressing ctrl and + at the same time
another solution would be to delete ~/.phy/TemplateGUI/state.json
Also, make sure that the waveforms are found on channel 0

sv-bio, 2020-12-28:
Dear Cyrille,

After closer look at the problem I found that it is always the first electrode in a set that has missing PCs. I described this issue in details at the forum of Spyking Circus, as I do not know to whom it should be addressed.
Please take a look: https://github.com/spyking-circus/spyking-circus/issues/336

Sincerely,
Stan