cortex-lab/phy #1066: Assertion issue, problem to load file with one electrode per group

Clementp69, 2021-01-17:
Hi everyone,

I am using SpikeInterface to sort spikes from my 16 channel recordings (200µm space between contact).

I've configured my probe file in order to have one channel per group.

This doing 16 sortings, one per channel. My sorting is okay but when I wanted to export it on phy with these commands :

st.postprocessing.export_to_phy(recording, sorting_IRON, output_folder='sorter140121', grouping_property='group')
!phy template-gui sorter140121/params.py

I have this error :

>11:58:55.853 [E] init:62 An error has occurred (AssertionError):
Traceback (most recent call last):
File "c:\program files\anaconda3\lib\runpy.py", line 193, in _run_module_as_main
"main", mod_spec)
File "c:\program files\anaconda3\lib\runpy.py", line 85, in run_code
exec(code, run_globals)
File "C:\Program Files\Anaconda3\Scripts\phy.exe_main.py", line 7, in
sys.exit(phycli())
File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 764, in call
return self.main(*args, **kwargs)
File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 717, in main
rv = self.invoke(ctx)
File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 1137, in invoke
return process_result(sub_ctx.command.invoke(sub_ctx))
File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 956, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "c:\program files\anaconda3\lib\site-packages\click\core.py", line 555, in invoke
return callback(*args, **kwargs)
File "c:\program files\anaconda3\lib\site-packages\click\decorators.py", line 17, in new_func
return f(get_current_context(), *args, **kwargs)
File "c:\program files\anaconda3\lib\site-packages\phy\apps_init.py", line 159, in cli_template_gui
template_gui(params_path, **kwargs)
File "c:\program files\anaconda3\lib\site-packages\phy\apps\template\gui.py", line 198, in template_gui
controller = TemplateController(model=load_model(params_path), dir_path=dir_path, **kwargs)
File "c:\program files\anaconda3\lib\site-packages\phylib\io\model.py", line 1178, in load_model
return TemplateModel(**get_template_params(params_path))
File "c:\program files\anaconda3\lib\site-packages\phylib\io\model.py", line 297, in init
self._load_data()
File "c:\program files\anaconda3\lib\site-packages\phylib\io\model.py", line 356, in _load_data
self.sparse_templates = self._load_templates()
File "c:\program files\anaconda3\lib\site-packages\phylib\io\model.py", line 652, in _load_templates
assert cols.shape == (n_templates, n_channels_loc)
AssertionError

Does someone have a solution to this?
Is it possible in Phy to load file with one channel per group ? 

Thanks in advance, 
Clément G

luciano-censoni, 2021-05-15:
Commenting because I have an identical issue.

zm711, 2024-06-07:
We have worked on the phy exporter in spikeinterface. If this is still an issue I would considering opening an issue in spikeinterface and opening a new issue here (linked) so we can track this.

closed 2024-06-07 by zm711