cortex-lab/phy #1068: "Issues" on MacOS 11

mschottdorf, 2021-01-22:
Hi all, 

I installed Phy on MacOS 11, and stumbled into an annoying error. As others might have experienced this problem as well,  I thought it could be useful to document the solution:

When trying to run phy, it throws the error:  `ImportError: ('Unable to load OpenGL library', 'dlopen(OpenGL, 10): image not found', 'OpenGL', None)`. This is not an issue with Phy per se; the same error happened when trying  in aa phyton window `from OpenGL import GL`.

The problem is a OpenGL complication of MacOS 11. pyOpenGL tries to find OpenGL via ctypes, and fails.

This is the solution:  One can edit the PyOpenGL file `OpenGL/platform/ctypesloader.py` and tell it exactly where openGL is located. i.e. replace line ` fullName = util.find_library( name )` with ` fullName = '/System/Library/Frameworks/OpenGL.framework/OpenGL' `

This solved the problem for me.

-Manuel

HammadFKhan, 2021-05-31:
Thank you for this tips. It works for me as well!

czuba, 2021-06-09:
Worked for me too. Thanks @mschottdorf 

Instance of ctypesloader.py found in my phy2 env (python3.7) included this half-workaround:
``` python
    if isinstance( dllType, ctypes.LibraryLoader ):
        dllType = dllType._dlltype
    if sys.platform.startswith('linux'):
        return _loadLibraryPosix(dllType, name, mode)
    else:
        return _loadLibraryWindows(dllType, name, mode)

def _loadLibraryPosix(dllType, name, mode):
    """Load a given library for posix systems

    The problem with util.find_library is that it does not respect linker runtime variables like
    LD_LIBRARY_PATH.
    # ...yadda yadda yadda...
    """
```
So it appears MacOS 11 users weren't the only ones getting stuck here.

Continuing to the `_loadLibraryWindows` method was also erroring on MacOS, so I just hacked in a `_loadLibraryMacOs` alternative following @mschottdorf's suggestion:
``` python
    if isinstance( dllType, ctypes.LibraryLoader ):
        dllType = dllType._dlltype
    if sys.platform.startswith('linux'):
        return _loadLibraryPosix(dllType, name, mode)
    elif sys.platform.startswith('darwin'):
        return _loadLibraryMacOs(dllType, name, mode)
    else:
        return _loadLibraryWindows(dllType, name, mode)
```
& 
``` python
def _loadLibraryMacOs(dllType, name, mode):
    """Load a given library for MacOS 11(+) systems with hardcoded OpenGL path

    returns the ctypes C-module object
    """
    name = '/System/Library/Frameworks/OpenGL.framework/OpenGL'
    try:
        return dllType( name, mode )
    except Exception as err:
        err.args += (name, 'MacOS_hardcoded')
        raise
```

...not that its particularly better/elegant/necessary, but in any case curious that something that seems so elemental would be hanging like this.

Elliott365, 2021-06-18:
I have a fresh install of phy2 and encountered the same error. When trying the solution listed by @mschottdorf I encountered this error 

ImportError: ('Unable to load OpenGL library', 'dlopen(/System/Library/Frameworks.OpenGl.framework/OpenGL, 10): image not found', '/System/Library/Frameworks.OpenGl.framework/OpenGL', '/System/Library/Frameworks.OpenGl.framework/OpenGL')

czuba, 2021-06-18:
@Elliott365 Check your string.
“Gl” ~= “GL” 🧐👍🏼

Elliott365, 2021-06-18:
much appreciated. Now I get:

Error: Invalid value for 'PARAMS_PATH': Path 'params.py' does not exist.

when running phy template-gui params.py

HammadFKhan, 2021-06-18:
Make sure that the terminal path leads to where you exported your kilosort
data. In that directory, kilosort has made a params.py file.

Also refer to the kilosort documentation on how to correctly output files.