cortex-lab/phy #1070: Phy failure to launch

verocg, 2021-02-03:
Hello,

I am having issues getting Phy operational. Kilosort is working fine, and generates all the correct files (.npy). However, when I try to launch Phy, I get a list of error messages. I am new to coding, and have no experience with Python, so even opening these files I haven't been able to figure out the issue.

The output from trying to launch "phy template-gui params.py" is below:

←[33m18:30:00.081 [W] model:545            Unreferenced clusters found in templates (generally not a problem)←[0m
←[33m18:30:00.084 [W] model:567            Unreferenced clusters found in spike_clusters (generally not a problem)←[0m
←[33m18:30:00.090 [W] model:54             104/225 values are nan in C:\Users\TDT\Documents\DATA\Dingo\02022021\RFmap_1\similar_templates.npy, replacing by zero.←[0m
←[0m18:30:00.187 [I] state:180            Copied c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\apps\template\static\state.json to C:\Users\TDT\.phy\TemplateGUI\state.json.←[0m
←[31m18:30:00.318 [E] __init__:62          An error has occurred (AssertionError):
Traceback (most recent call last):
  File "c:\users\tdt\anaconda3\envs\phy2\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\TDT\anaconda3\envs\phy2\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 1025, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 955, in main
    rv = self.invoke(ctx)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 1517, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 1279, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\click\core.py", line 710, in invoke
    return callback(*args, **kwargs)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\click\decorators.py", line 18, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\apps\template\gui.py", line 199, in template_gui
    gui = controller.create_gui()
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 1638, in create_gui
    self.supervisor.attach(gui)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\cluster\supervisor.py", line 942, in attach
    gui=gui, sort=gui.state.get('ClusterView', {}).get('current_sort', None))
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\cluster\supervisor.py", line 760, in _create_views
    gui, data=self.cluster_info, columns=self.columns, sort=sort)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\cluster\supervisor.py", line 916, in cluster_info
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\cluster\supervisor.py", line 916, in <listcomp>
    return [self.get_cluster_info(cluster_id) for cluster_id in self.clustering.cluster_ids]
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\cluster\supervisor.py", line 745, in get_cluster_info
    out[key] = func(cluster_id)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 1150, in get_best_channel_label
    return self._get_channel_labels([self.get_best_channel(cluster_id)])[0]
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\utils\context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 1144, in get_best_channel
    channel_ids = self.get_best_channels(cluster_id)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\utils\context.py", line 154, in memcached
    out = f(*args, **kwargs)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phy\apps\template\gui.py", line 149, in get_best_channels
    template = self.model.get_template(template_id)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 855, in get_template
    return self._get_template_dense(template_id, channel_ids=channel_ids)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 798, in _get_template_dense
    channel_ids_, amplitude, best_channel = self._find_best_channels(template)
  File "c:\users\tdt\anaconda3\envs\phy2\lib\site-packages\phylib\io\model.py", line 787, in _find_best_channels
    assert best_channel in channel_ids
AssertionError
←[0m

closed 2021-02-17 by verocg

neurvanna, 2021-03-05:
I'm having the same issue, could you please share how you've solved this? Thanks!

verocg, 2021-03-08:
I think my issue was due to kilosort not saving all the output files correctly due to access issues to my output folder. When I changed the output directory to a different folder with full write access this issue with phy went away. Hope this helps if you haven't already resolved your issue!