cortex-lab/phy #1071: Phy is not able to generate FeatureView nor TemplateFeatureView in its GUI

tobe2098, 2021-02-04:
The error stands : "15:03:52.934 [W] gui:671              Could not create view FeatureView" and "15:03:54.478 [W] gui:671              Could not create view TemplateFeatureView". Resetting with --clear-state or --clear-cache does not solve the issue. I will post any additional information that you might need on request.

alejandrotorrado, 2021-02-05:
I just set up Phy yesterday and am running into the same issue.

tobe2098, 2021-02-08:
Update: Seems that the problem is Kilosort related, sessions that used K2.5 had no problem but those that used K3.0 had this issue

JudithLim, 2021-02-12:
I'm having the same issue when using the Kiwk-Gui on spiking data sorted with Klusta, so it might not necessarily be KiloSort related. In case it might be handy, I get the following output:
```
Traceback (most recent call last):
  File "c:\users\judith\anaconda3\envs\phy2\lib\site-packages\phy\gui\qt.py", line 532, in run
    result = self.fn(*self.args, **self.kwargs)
  File "c:\users\judith\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 171, in _worker
    self.on_select(cluster_ids=cluster_ids, **kwargs)
  File "c:\users\judith\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\base.py", line 142, in on_select
    self.plot(**kwargs)
  File "c:\users\judith\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 337, in plot
    self._plot_cluster(bunch)
  File "c:\users\judith\anaconda3\envs\phy2\lib\site-packages\phy\cluster\views\waveform.py", line 212, in _plot_cluster
    assert masks.shape == (n_spikes_clu, n_channels)
AssertionError
[33m11:17:08.736 [W] gui:671              Could not create view FeatureView.[0m
```

m-col, 2021-02-16:
@tobe2098 I did a bit of debugging and we are hitting https://github.com/cortex-lab/phy/blob/master/phy/apps/base.py#L391, blocking the creation of the feature view. It seems our setup (sorting with kilosort via spikeinterface) is causing kilosort not to output the files needed for phylib to create feature view (i.e. not 'pc_features.npy'). It is likely the spikeinterface code that handles kilosort 3 needs a bit of ironing out as it's very new.

This seems to be a different error to what you are seeing @JudithLim.

m-col, 2021-02-16:
Looks like it has been discussed on the kilosort issue tracker -- https://github.com/MouseLand/Kilosort/issues/317 -- and will be back at some point.

JudithLim, 2021-02-18:
@m-col  Yea, you're probably right. I can't seem to get klustakwik2 properly installed within the phy2 environment, so that's probably where the error is coming from.

MiguelVillavicencio, 2021-04-01:
Hello All,
I introduced myself with Kilosort by using the version 3 and had the same problem (No Feature/Template Views). So guided by the comments here I switched to Kilosort 2.5 and tried but this time phy crashes and does not even run/open the GUI. I don't know if one version 3 configurations prevents version 2.5 to open or if my phy is 'tied' to one of those versions, do you have any suggestion in this regard? (I am kind of stuck: I cannot finalize my sorting without the Feature view and I cannot use K2.5 either)

jamespherman, 2022-03-02:
Bump - what's going on with this issue?

Zoe0793, 2022-04-14:
A temporary work around has been stated here: https://github.com/MouseLand/Kilosort/issues/429

Aghileh, 2022-07-12:
Thanks for your reply and explanation.
I checked, seems that cProj, iNeigh, iNeghPC, cProjPC is empty in rez file. should I find the line that they overwrite as empty?! or are they other fields that you checked before?!

Aghileh, 2022-07-12:
> I was able to get FeatureView back in 2.5 by uncommenting / commenting-out a bunch of lines of code. I tried doing the same thing in KS3 and could not make it work.
> 
> I don't have the code handy right now so I can't give you all the information, but I can describe the process I used to identify which lines to uncomment: I noticed that the npy/csv files you mentioned above were not being generated because the variables that are stored in the files are empty when KS tries to write them. I worked backwards from the write point, setting breakpoints at progressively earlier spots in the code and discovered that at some point the variables are not empty, but they are overwritten with empty ("[]") to avoid carrying them around (there are some comments about wanting to avoid holding them in memory, iirc). I commented out the lines of code where the variables were overwritten with empty, and the files were written.
> 
> I hope this helps, and good luck! I hope FeatureView comes back, I consider KS unusable without it, personally.

Thanks for your reply and explanation.
I checked, seems that cProj, iNeigh, iNeghPC, cProjPC is empty in rez file. should I find the line that they overwrite as empty?! or are they other fields that you checked before?!

AHSAN-PA, 2024-05-07:
We can fix FeatureView. Please refer https://github.com/cortex-lab/phy/issues/1267 and https://github.com/cortex-lab/phy/issues/1145.