cortex-lab/phy #1074: Error when executing action next

m-beau, 2021-02-14:
System:
Ubuntu 20.04
dataset on a SATA-connected 4TB SSD
nvidia quadro GPU
32GB RAM

Steps to reproduce the problem:
Open phy2, select a unit, hit space

Full error log:
```
18:22:47.249 [E] qt:178               Timeout in _block().
18:23:24.407 [W] actions:213          Error when executing action Next.
18:23:24.509 [D] actions:214          Traceback (most recent call last):
  File "/home/maxime/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/gui/actions.py", line 211, in wrapped
    return callback(*args)
  File "/home/maxime/anaconda3/envs/phy2/lib/python3.7/site-packages/phylib/utils/event.py", line 141, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/maxime/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/cluster/supervisor.py", line 847, in _on_action
    _block(lambda: not self._is_busy)
  File "/home/maxime/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/gui/qt.py", line 182, in _block
    raise RuntimeError("Timeout in _block().")
RuntimeError: Timeout in _block().
```

HadarBeeri, 2021-02-15:
I also tried to work with phy2 today , and get the same error.

m-beau, 2021-02-17:
Phy also regularly times out when trying to perform a split. It is a performance issue rather than a bug.

reinderdorman, 2021-02-19:
Have the same error, however, there is no traceback printed in the terminal what receiving this error. It does happen on moving to 'next best'.

Seems to happen on clustered data from one recording, clustering of two seperate probes from this recording., (have seen the error before, but not as abundant). Tested a random other recording and the error didn't occur. I can provide this affected dataset if needed.

Notably, the affected data have quite a lot of clusters (200+, which for my data is quite a bit.)

Additionally, the gui seems to be a bit slower than normal, but I'm not sure of this. Many times the waveform viewer is lagging behind when a new cluster is selected, and the mouse cursus disappears during this moment. However, the mouse is still around, as you can click on things inside the gui.

Have tried uninstalling and reinstalling the entire phy environment, to no avail.

On Ubuntu 16.04, intel i7-6700K, 32GB ram. Resource monitor does not show any RAM or CPU bottleneck of sorts. 

(might I add that the affected sessions had beeeeeaaaaauuutiful cells, so I definitely want to salvage this one!)

arthurz323, 2021-04-29:
same issue happened to me when i tried to execute action, with 700+ clusters.