cortex-lab/phy #1078: segmentation fault, seemingly random on multiple datasetes

reinderdorman, 2021-02-22:
Hi all,

I'm getting Segmentation Fault errors. I can't find a pattern when it happens. Happened on multiple datasets so far, rather sparse (i.e. on some occasions just once or twice a day?).

Using a fresh install (last week) of phy2 (version 2,0b1), on Ubuntu 16.04.

Would love to share the datasets which show this error, but I have a feeling this might be a difficult to reproduce issue. Nonetheless, I wanted to mention it.

reinderdorman, 2021-11-30:
This problem still pops up from time to time. Today it happened while simply trying to select a different cluster. To help out I have now copied the entire error traceback:

17:00:14.955 [E] __init__:62          An error has occurred (AssertionError): 
Traceback (most recent call last):
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/gui/widgets.py", line 260, in emitJS
    self._debouncer.submit(emit, *args)
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/gui/qt.py", line 605, in submit
    self.trigger()
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/gui/qt.py", line 620, in trigger
    f(*args, **kwargs)
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phylib/utils/event.py", line 141, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/cluster/supervisor.py", line 821, in _clusters_selected
    emit('select', self, self.selected, **kwargs)
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phylib/utils/event.py", line 141, in emit
    res.append(f(sender, *args, **kwargs))
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/base.py", line 315, in on_select
    view.plot()
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/cluster/views/amplitude.py", line 225, in plot
    self.canvas.update_visual(self.visual)
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/plot/plot.py", line 142, in update_visual
    data = visual.set_data(*args, **kwargs) if kwargs else None
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/plot/visuals.py", line 220, in set_data
    data = self.validate(*args, **kwargs)
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/plot/visuals.py", line 211, in validate
    data_bounds = _get_data_bounds(data_bounds, pos)
  File "/home/ephys/anaconda3/envs/phy2/lib/python3.7/site-packages/phylib/utils/geometry.py", line 203, in _get_data_bounds
    assert data_bounds.shape == (length, 4)
AssertionError

Segmentation fault (core dumped)

sv-bio, 2022-01-10:
Hi,

Same error. Phy2 (version 2,0b1), Debian 10

Traceback (most recent call last):
  File "/path/user/miniconda2/envs/circus/lib/python3.6/site-packages/phy/gui/qt.py", line 532, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/user/miniconda2/envs/circus/lib/python3.6/site-packages/phy/cluster/views/base.py", line 175, in _worker
    self.on_select(cluster_ids=cluster_ids, **kwargs)

a-zmz, 2024-01-12:
got the same error on all my datasets:
`[1]    15616 segmentation fault (core dumped)  env QTWEBENGINE_CHROMIUM_FLAGS='--single-process' phy template-gui params.py`