cortex-lab/phy #1082: Problem with WaveformView

FraCost, 2021-02-26:
WaveformView is showing flattened spike waveforms, how can I fix it?
[image]

claudiagjou, 2021-03-04:
I have the same issue, but only if I change n spikes in waveformView to anything other than 200 spikes.

epmgs97, 2021-05-20:
I have the same issue too... Try to 'increase' with **ctrl + Up**, you may then see your waveforms, but careful cause the other views will change too! It might be a problem with the filter scale reajustment that Kilososrt does when it takes our data to preprocess it. Maybe our data is in a different scale than the one used by Kilosort to represent the filtered waveform, so then when Phy tries to painted it seems like no wave is been captured, even though Kilosort does identify the spike. 

Does anyone knows the range of values Kilosort wants to be given the raw data in the .bin file so the filter could finally be well visuelized in Phy?

GrimmSnark, 2022-04-14:
So I have found a overkill solution to this, at least for Windows. Whenever you run phy it saves a hidden folder with all the memcache and various GUI settings within the data folder. This folder is named ".phy", delete that and it seems to reset the waveform amplitude disparity.

Michael