cortex-lab/phy #1086: Waveformview- toggle mean waveforms does not work

JannaAarse, 2021-03-16:
Hey, 

I am using phy2 to curate my data (which is sorted with kilosort2.5) and I try to switch between the template view and the mean waveform view to make sure my curation is good but it does not change. It doesn't matter if I click on it on the menu or if I press "M" nothing changes. I can zoom in and out so the window is not frozen, I can even switch from template_high_pass to template_raw when I press "alt+r" but no matter what I do I don't get the mean_waveform. I thought it might be that I am not up-to-date with my phy version but I did run an update (pip install phy --pre --upgrade) but that also doesn't change anything. So I went I tried the same kilosort file on a different computer and it also doesn't show up there. This does happen for a lot of different data sets. Do you have an idea what else I can try to get this fixed ? Or did you saw this before and I am just missing something ? Or maybe it doesn't have anything to do with phy but rather with a step before ? 

Best, 
Janna

rclein94, 2022-04-08:
Hi I'm wondering if you ever resolved this issue? I'm experiencing a similar issue. thank you!

GrimmSnark, 2022-04-14:
So I have found a overkill solution to this, at least for Windows. Whenever you run phy it saves a hidden folder with all the memcache and various GUI settings within the data folder. This folder is named ".phy", delete that and it seems to reset the waveform amplitude disparity. 

Michael

3man1992, 2023-06-06:
The above solution didn't work for me. What did was:

`phy template-gui params.py --clear-state`