cortex-lab/phy #1091: only one waveform being displayed per cluster in WaveformView

atranvan, 2021-04-12:
Hi, 

I am using phy (v2.0a1) to visualize data analyzed with klusta. OS is Ubuntu 20.04. 
No matter the value I set n_spikes_waveforms to, only 1 waveform per cluster gets displayed in WaveformView. I have re-opened old files which now do show the same issue (and did not use to). Any clue what I may have set wrong or how to reset WaveformView parameters?

thanks in advance

Tywang-720, 2021-04-29:
It's showing mean waveform I guess.

Normally we could switch between mean waveform view and all waveform view by pressing 'M', or click "toggle mean waveform view" in the menu of waveform view window.

But sometimes it just won't work.

jluo2340, 2021-05-04:
I have this issue as well. Pressing 'M' and "Toggle Mean Waveform" doesn't work anymore for my files. I've tried running phy with --clear-state and --clear-cache as well, and I'm still unable to see more than 1 waveform.

atranvan, 2021-05-04:
Hi,

toggling M didn't work for me either. It wouldn't explain why some channels showed an empty line, since the mean would still be a noisy trace
In the end what I did is:

- install the last version of phy  in a fresh environment
- the WaveformView window still displayed only 1 trace, as before
- but I could now create a new WaveformView window that displayed all waveforms properly, and delete the old ones

not sure it works for anyone else, but @jluo2340 it may be worth a shot?

atranvan, 2021-05-04:
sorry I realized I had not mentionned the empty line issue earlier. Not only the visualization contained one trace per channel only, I also had several completely flat lines - hence not just a "Toggle Mean Waveform" issue

eabhorrocks, 2021-08-09:
Have the same issue, including when opening a new Waveform View window. Data is output of kilosort3

rclein94, 2022-04-08:
Hi, I'm having the same issue, wondering if this was ever resolved. If it was, can you please share tips? Thanks!

eintzandt, 2022-12-05:
I run into this bug often, was wondering if there was any update? I have tried deleting the .phy config folder and that doesn't work.

GoktugAlkan, 2023-06-06:
Same trouble. I am using the latest version of phy and trying to visualize waveforms obtained with Kilosort 2.5. It just displays the mean waveform. When I use the same phy with the outputs obtained with Kilosort 1.0 it works without any problems. Is there an update?