cortex-lab/phy #1096: Co-Curing multiple sessions worth of data

achristensen56, 2021-05-13:
Hi!

I was wondering if anyone has any cute tricks as far as trying to use Phy to cure data that has been clustered across multiple sessions. For example, is there an easy way to add some metadata such that waveforms, clusters, etc. from session1 (defined by spike timestamp, e.g.) are displayed in a different color than the data from session2? I'm looking at the customization section of the docs and considering writing a widget for this, but before I dive in I was wondering if there are easy existing ways to use phy for this purpose.

what would be great is if we could have two windows, similar to clusters and similar clusters, which corresponded to either two separately sorted datasets, *or* that corresponded to two separate *time spans* of a single sorted dataset. 

thanks!
Amy

czuba, 2021-06-15:
Commenting/subscribing to say that this is something I would also love to see developed & would be very useful; either as a plugin or full-fledged multi-session feature in Phy.  ...particularly, as you mentioned, for files that require running split subsets to be run through kilosort, then post-hoc merged manually —e.g. files with an abrupt spatial (probe shift) or temporal (time gap) discontinuity.

@achristensen56 curious if you've tried/succeeded in manually concatenating the output files from separate kilosort sessions, then loading the merged set into Phy?  It wouldn't be as pretty as the true Phy extension you've proposed, but the session assignment should be obvious in the amplitude view.

[...even a simple  _"nah, didn't try it"_  or  _"yeah, but it was a nightmare rabbit-hole"_  would be much appreciated.]

---

In either case, in addition to combining clusters with [subjectively] coherent waveforms from different sorting sessions, this kind of 'merged Phy session' would be great for pooling low-threshold clusters from a local region (channel, adjacent pair, row of sites, whatever) into a common multi-unit cluster across sorting sessions.

...the sort of cluster merging that traditionally occurred during hand-sorting spikes into a _distinctly-not-noise—but-also-not-separable-clusters_ MUA group that would otherwise fall through the cracks when applying a post-hoc "mergeSessionSingleUnits" function that only considered mean waveforms for matching clusters across sorting sessions.

achristensen56, 2021-06-15:
I actually never got around to implementing that! We decided to cure the individual days from session1 and session2 then match neurons in the co-sorted session, accepting a neuron if it was accepted in both constituent days. 

But also curious if you try the concatenation how it goes!