cortex-lab/phy #1107: problem opening kilosort files when threshold is bellow 10 (for example [9 4])

yasminya, 2021-08-17:
Hi,
I'm using kilosort 2.5 followed by phy. For some files I ran kilosort with different threshold combinations. All results files with threshold <10 can't be opened in Phy (When running kilosort again with threshold 10 everything works fine). Can anyone explain why this happens? With threshold 10 kilosort misses many of the spikes in these files
thanks,

rossant, 2021-09-07:
How many spikes are there in the files that phy can't open?

yasminya, 2021-09-09:
I ran kilosort again now and things changed a bit. For threshold [10 4] and [9 4] i got 164 spikes and 193 spikes, respectively, and can open the files in phy. For threshold [6 4] i got 361 spikes and file doesn't open in phy.