cortex-lab/phy #1110: ImportError: cannot import name 'SpikeSelector' from 'phylib.io.array'

yoonbai, 2021-08-30:
I get this error after re-installing conda & phy. I tried changing the 'Selector' class definition (phylib.io.array) to 'SpikeSelector' but ran into additional issues. Perhaps there is a mismatch b/w phy & phylib versions?

I'm seeing similar issues under a [closed thread](https://github.com/cortex-lab/phy/issues/1000 ) as well.

Thanks for looking into this!

mariehemelt, 2021-08-30:
I am also having this problem. Thanks for any ideas!

meszena, 2021-08-31:
I have the same situation. But there are active commits going on at the moment.
I think it will be resolved soon, it should be just some version conflict.

meszena, 2021-09-02:
**ImportError: DLL load failed** "
Oops, now I've got another error, did it change for you too? " 

  _File "c:\users\mesze\.conda\envs\phy2\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\mesze\.conda\envs\phy2\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\mesze\.conda\envs\phy2\Scripts\phy.exe\__main__.py", line 4, in <module>
  File "c:\users\mesze\.conda\envs\phy2\lib\site-packages\phy\apps\__init__.py", line 22, in <module>
    from phy.gui.qt import QtDialogLogger
  File "c:\users\mesze\.conda\envs\phy2\lib\site-packages\phy\gui\__init__.py", line 6, in <module>
    from .qt import (
  File "c:\users\mesze\.conda\envs\phy2\lib\site-packages\phy\gui\qt.py", line 31, in <module>
    from PyQt5.QtCore import (Qt, QByteArray, QMetaObject, QObject,  # noqa
ImportError: DLL load failed: The specified procedure could not be found._

mariehemelt, 2021-09-03:
No, mine still looks the same today: 

(phy2) C:\Users\mh553\TempFromIsilonQ\TotalSort>phy template-gui params.py
Traceback (most recent call last):
  File "c:\users\mh553\anaconda3\envs\phy2\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\mh553\anaconda3\envs\phy2\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\mh553\Anaconda3\envs\phy2\Scripts\phy.exe\__main__.py", line 4, in <module>
  File "c:\users\mh553\anaconda3\envs\phy2\lib\site-packages\phy\apps\__init__.py", line 25, in <module>
    from .base import (  # noqa
  File "c:\users\mh553\anaconda3\envs\phy2\lib\site-packages\phy\apps\base.py", line 22, in <module>
    from phylib.io.array import Selector, _flatten
ImportError: cannot import name 'Selector' from 'phylib.io.array' (c:\users\mh553\anaconda3\envs\phy2\lib\site-packages\phylib\io\array.py)

juliecharlton, 2021-09-03:
I still get the same error (mine is for 'Selector' instead of 'spikeSelector'

_Traceback (most recent call last):
  File "/Users/gorislab/opt/anaconda3/envs/phy2/bin/phy", line 5, in <module>
    from phy.apps import phycli
  File "/Users/gorislab/opt/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/__init__.py", line 25, in <module>
    from .base import (  # noqa
  File "/Users/gorislab/opt/anaconda3/envs/phy2/lib/python3.7/site-packages/phy/apps/base.py", line 22, in <module>
    from phylib.io.array import Selector, _flatten
ImportError: cannot import name 'Selector' from 'phylib.io.array' (/Users/gorislab/opt/anaconda3/envs/phy2/lib/python3.7/site-packages/phylib/io/array.py)_

juliecharlton, 2021-09-03:
I found a very hacky solution that seems to work. I went into phy2/lib/python3.7/site-packages/phy and manually replaced all of the code that's been updated recently (<2 years old) with the code from github.

yoonbai, 2021-09-03:
LOL Thanks, Julie! Hope phy & phylib branches sync up again.

Feikai-Lin, 2021-09-06:
> I found a very hacky solution that seems to work. I went into phy2/lib/python3.7/site-packages/phy and manually replaced all of the code that's been updated recently (<2 years old) with the code from github.

This is helpful! I download phy-master from github and replaced the original phy (phy2/lib/python3.7/site-packages/phy) with the updated phy from phy-master. It seems to solve this issue.

rossant, 2021-09-07:
Hopefully this should be fixed with the latest phy beta release, feel free to reopen if not

closed 2021-09-07 by rossant